Drewsif/PiShrink #1: [Feature Request] Add a test option

Septolum, 2016-04-26:
Add a test option, that does everything and outputs the new size of the .img file, but doesn't overwrite the original file
For compatibility with testing files on read only filesystems and for paranoid users

Drewsif, 2016-04-27:
Writing out to a different file I don't think can be done since the programs that shrink the filesystem are expecting a block device that they can live edit. 

The issue is similar with a test option, its possible to approximate how much space could be saved but there is no real way to test the resizing without actually doing it. I'm open to ideas if you think I overlooked something though.

-Drew

Drewsif, 2016-04-28:
While not quite the most efficient way, you can at least have a backup made before it does things

closed 2016-04-28 by Drewsif