Drewsif/PiShrink #4: Not working on Raspbian image

XP1, 2016-07-06:
Not working on Raspbian image

```
sudo ./pishrink.sh 2016-05-27-raspbian-jessie.img new.img
Copying 2016-05-27-raspbian-jessie.img to new.img...
Creating new /etc/rc.local
e2fsck 1.42.13 (17-May-2015)
Pass 1: Checking inodes, blocks, and sizes
Pass 2: Checking directory structure
Pass 3: Checking directory connectivity
Pass 4: Checking reference counts
Pass 5: Checking group summary information
/dev/loop0: 129993/241440 files (0.1% non-contiguous), 856512/964096 blocks
resize2fs 1.42.13 (17-May-2015)
resize2fs 1.42.13 (17-May-2015)
The containing partition (or device) is only 964096 (4k) blocks.
You requested a new size of 984096 blocks.

ERROR: resize2fs failed...
```

Raspbian download:
https://www.raspberrypi.org/downloads/raspbian/

XP1, 2016-07-06:
I was able to shrink the Raspbian image with these commands:

`cp 2016-05-27-raspbian-jessie.img test.img`

Look for the start of the partition, which is 137216:

```
fdisk -l test.img

Disk test.img: 3.8 GiB, 4019191808 bytes, 7849984 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x14c20151

Device     Boot  Start     End Sectors  Size Id Type
test.img1         8192  137215  129024   63M  c W95 FAT32 (LBA)
test.img2       137216 7849983 7712768  3.7G 83 Linux
```

Mount partition at 137216:

`losetup --offset $((137216 * 512)) /dev/loop0 test.img`

Look at the number of blocks ("855501/912896 blocks"):

```
e2fsck -f /dev/loop0

e2fsck 1.42.13 (17-May-2015)
Pass 1: Checking inodes, blocks, and sizes
Pass 2: Checking directory structure
Pass 3: Checking directory connectivity
Pass 4: Checking reference counts
Pass 5: Checking group summary information
/dev/loop0: 129992/225344 files (0.1% non-contiguous), 855501/912896 blocks
```

Pick a number of blocks above 855501 blocks. I picked 900000 blocks.

```
resize2fs -f /dev/loop0 900000
e2fsck -f /dev/loop0
```

Unmount:

`losetup -d /dev/loop0`

Run the script:

```
chmod +x pishrink.sh
./pishrink.sh test.img 
Creating new /etc/rc.local
e2fsck 1.42.13 (17-May-2015)
Pass 1: Checking inodes, blocks, and sizes
Pass 2: Checking directory structure
Pass 3: Checking directory connectivity
Pass 4: Checking reference counts
Pass 5: Checking group summary information
/dev/loop0: 129993/225344 files (0.1% non-contiguous), 855502/900000 blocks
resize2fs 1.42.13 (17-May-2015)
resize2fs 1.42.13 (17-May-2015)
Resizing the filesystem on /dev/loop0 to 920000 (4k) blocks.
The filesystem on /dev/loop0 is now 920000 (4k) blocks long.

Shrunk test.img from 3.8G to 3.6G
```

It works, but I only have 20 MB free space left with a 4 GB SDHC card.

```
df -h

Filesystem      Size  Used Avail Use% Mounted on
/dev/root       3.5G  3.3G   20M 100% /
devtmpfs        214M     0  214M   0% /dev
tmpfs           218M     0  218M   0% /dev/shm
tmpfs           218M  4.6M  213M   3% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           218M     0  218M   0% /sys/fs/cgroup
/dev/mmcblk0p1   63M   21M   43M  33% /boot
tmpfs            44M     0   44M   0% /run/user/1000
```

So now to uninstall programs to free some space...

Drewsif, 2016-07-07:
Thanks for reporting the issue. The script was trying to shrink the filesystem to a size larger than the actual filesystem. Pushing a patch for it now!

closed 2016-07-07 by Drewsif