Drewsif/PiShrink #6: rc.local won't autoexecute, system is not auto-expanded at 1st boot!

aikoncwd, 2016-07-21:
Hi, I used your tool months ago and it worked like a charm. Today I shrinked a new image, everything looks ok to me (no errors on console while using your tool).

The problem is, when I write the 4.4Gb-shrinked-image to my 32Gb-microSD... the system boots ok, but the content of `/etc/rc.local` is not executed, so the system is not auto-expanded, also the original rc.local is not replaced and there is no reboot at the end.

If I reboot again the system, rc.local still won't auto-execute, BUT if I run manually `/etc/rc.local` then all the commands are executed, system is expanded, rc.local is restored and finally the system goes to reboot.

Can you help me with that? I can send you more information, also I can send you my *.img file so you can test that issue! Thanks :)

Drewsif, 2016-08-07:
Hmm maybe your system didn't have rc.local enabled? Which pi image are you running?

Drewsif, 2016-08-18:
Pull #7 hopefully fixed this issue. If not please let me know!

closed 2016-08-18 by Drewsif

aikoncwd, 2016-08-19:
Yep, that was causing the issue. It's now fixed :smile: