Drewsif/PiShrink #10: "ERROR: You need to be running as root." on Windows

NeonNegro, 2016-11-25:
when I delet  this "IF" block:
if (( EUID != 0 )); then
  echo "ERROR: You need to be running as root."
  exit -3
fi

I get the error: ERROR: parted is not installed

Drewsif, 2016-11-26:
How are you running it on Windows? I made and tested it on Linux so that why it won't run. You can shrink it from the pi though

NeonNegro, 2016-11-26:
Didn't know it was only for windows, sorry. Now that I know, two questions: it is possible to make a windows version one, and how can I run it from the Pi? I can't put the img inside the SD, there isn't enough space

Drewsif, 2016-11-28:
Its possible, but I would need a windows alternative to parted and the auto resize wouldn't really work. If you have a usb hard drive you can put the image on you can use it from the pi

closed 2016-11-28 by Drewsif