Drewsif/PiShrink #12: waitting at "part2=`parted $img unit B mkpart primary $partstart $newpartend`"

zhilinwww, 2017-06-09:
I add some comments in pishrink.sh, as follow:

#Shrink partition
losetup -d $loopback
echo disatch dev
part3=`parted $img print`
partnewsize=`expr $minsize \* $blocksize | tr -d '\n'`
newpartend=`expr $partstart + $partnewsize | tr -d '\n'`
echo cal partsize

part1=`parted $img rm $partnum`
part3=`parted $img print`

echo $part1 delete part

echo  $img $partstart $newpartend
part2=`parted $img unit B mkpart primary $partstart $newpartend`

echo finish Shrink partition

my running result is :

yhzk@yhzk-B85M-D3H:~/PiShrink-master$ sudo ./pishrink.sh webmin-jessie.img 
[sudo] password for yhzk: 
Creating new /etc/rc.local
e2fsck 1.42.9 (4-Feb-2014)
Pass 1: Checking inodes, blocks, and sizes
Pass 2: Checking directory structure
Pass 3: Checking directory connectivity
Pass 4: Checking reference counts
Pass 5: Checking group summary information
/dev/loop0: 188860/300144 files (0.2% non-contiguous), 1167002/1187107 blocks
resize2fs 1.42.9 (4-Feb-2014)
resize2fs 1.42.9 (4-Feb-2014)
Resizing the filesystem on /dev/loop0 to 1181653 (4k) blocks.
Begin pass 2 (max = 4637)
Relocating blocks             XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
Begin pass 3 (max = 37)
Scanning inode table          XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
The filesystem on /dev/loop0 is now 1181653 blocks long.

finish Shrink filesystem
disatch dev
cal partsize
delete part
webmin-jessie.img 48570880 4888621568

pishrink always waitting at "part2=`parted $img unit B mkpart primary $partstart $newpartend`"

The command is correct,but it never exit.
It trouble me for a long time . I'm looking forward to your reply.

Drewsif, 2017-06-18:
What OS were you using to run the script? And can you describe how you made the image you were shrinking so I can attempt to replicate the issue.

Drewsif, 2017-07-06:
Going to close it for now until I can replicate the issue

closed 2017-07-06 by Drewsif