Drewsif/PiShrink #13: Beware of the e2fsprogs version

vpetersson, 2017-07-04:
If you're getting the error:

```
[...]
has unsupported feature(s): metadata_csum.
[...]
```
This means that you're running on a too old version of e2fsprogs.

For instance, in order to run this on Ubuntu 16.04 (and earlier versions), you will need to compile e2fsprogs by hand and update the references in the script.  (Ubuntu comes with e2fsprogs version 1.42.13, but I was only able to get this working with 1.43.4.)

Drewsif, 2017-07-06:
Thanks for the heads up! I'll add something to the readme

raspberrycoulis, 2017-07-11:
I've been struggling with this issue too - however, I cannot get this working and I've tried various versions of Ubuntu (12.04, 14.04, 16.04) with no luck. Any guidance on how to update e2fsprogs would be greatly appreciated @vpetersson.

raspberrycoulis, 2017-07-12:
Ok, I have an update which I hope will help others if experiencing this issue. 

Simply use Ubuntu 16.10 or above and it works fine. I've tried 12.04, 14.04 and 16.04 and the issue about `has unsupported feature(s): metadata_csum.` is present, but using Ubuntu 16.10 and 17.04 works just fine.

Might be worth mentioning this in the README.md to include advice on this.

vpetersson, 2017-07-12:
@raspberrycoulis that's probably the path of least resistance. In short, what I did was to:

 * Compile e2fsprogs from source
 * Update the bash script and hard code the path to the binaries (from the compile above)

Drewsif, 2017-07-13:
@vpetersson is 1.43.4 the earliest working version or just the one you tested on?

vpetersson, 2017-07-14:
Yeah,  I used 1.43.4. 

Just for the record, here are my changes I made (obviously modify the path accordingly):

```
$ diff --git a/pishrink.sh b/pishrink.sh
index f2df6b9..3369eb8 100755
--- a/pishrink.sh
+++ b/pishrink.sh
@@ -1,4 +1,8 @@
-#!/bin/bash
+#!/bin/bash -ex
+
+TUNE2FS="/home/mvip/src/e2fsprogs-1.43.4/misc/tune2fs"
+RESIZE2FS="/home/mvip/src/e2fsprogs-1.43.4/resize/resize2fs"
+E2FSCK="/home/mvip/src/e2fsprogs-1.43.4/e2fsck/e2fsck"

 usage() { echo "Usage: $0 [-s] imagefile.img [newimagefile.img]"; exit -1; }

@@ -51,8 +55,8 @@ beforesize=`ls -lah $img | cut -d ' ' -f 5`
 partnum=`parted -m $img unit B print | tail -n 1 | cut -d ':' -f 1 | tr -d '\n'`
 partstart=`parted -m $img unit B print | tail -n 1 | cut -d ':' -f 2 | tr -d 'B\n'`
 loopback=`losetup -f --show -o $partstart $img`
-currentsize=`tune2fs -l $loopback | grep 'Block count' | tr -d ' ' | cut -d ':' -f 2 | tr -d '\n'`
-blocksize=`tune2fs -l $loopback | grep 'Block size' | tr -d ' ' | cut -d ':' -f 2 | tr -d '\n'`
+currentsize=`$TUNE2FS -l $loopback | grep 'Block count' | tr -d ' ' | cut -d ':' -f 2 | tr -d '\n'`
+blocksize=`$TUNE2FS -l $loopback | grep 'Block size' | tr -d ' ' | cut -d ':' -f 2 | tr -d '\n'`

 #Check if we should make pi expand rootfs on next boot
 if [ "$should_skip_autoexpand" = false ]; then
@@ -114,8 +118,8 @@ else
 fi

 #Make sure filesystem is ok
-e2fsck -f $loopback
-minsize=`resize2fs -P $loopback | cut -d ':' -f 2 | tr -d ' ' | tr -d '\n'`
+$E2FSCK -f $loopback
+minsize=`$RESIZE2FS -P $loopback | cut -d ':' -f 2 | tr -d ' ' | tr -d '\n'`
 if [[ $currentsize -eq $minsize ]]; then
   echo ERROR: Image already shrunk to smallest size
   exit -6
@@ -131,7 +135,7 @@ elif [[ `expr $currentsize - $minsize - 100` -gt 0 ]]; then
 fi

 #Shrink filesystem
-resize2fs -p $loopback $minsize
+$RESIZE2FS -p $loopback $minsize
 if [[ $? != 0 ]]; then
   echo ERROR: resize2fs failed...
   mount $loopback $mountdir
```

closed 2017-10-24 by Drewsif