Drewsif/PiShrink #14: How to Use

JourdanDixon, 2017-07-06:
Hi there, sorry to be a bit dense here.  Can I run this on my Mac?  The Micro SD for the Pi is 119 Gigs (I hate that they advertise as 128 Gigs and measure at 1000 Mb) and my image file is just under 128 Gigs.  So having 2 exist on the Pi wouldn't work.  Any suggestions are appreciated.  Thanks!

Drewsif, 2017-07-06:
No, you need to run it under Linux. I don't have a Mac to tests or develop against. If you have an external drive to put your image on you can use your pi to run the script.

closed 2017-07-06 by Drewsif

JourdanDixon, 2017-07-15:
So, I've placed the pishrink on the desktop of the pi and I have the image on an external hard drive.  I tried the below script.  Any suggestions?  It says Permission denied.  If I use sudo in front of it, it says command not found.  I tried [] around the img file source as well, but no change.

/home/pi/Desktop/pishrink.sh [-s] /media/pi/My Book/RetroPie Image/NoInputs128g_/NoInputs128g.img [/media/pi/My Book/RetroPie Image/NoInputs128g_/NewNoInputs128g.img]

JourdanDixon, 2017-07-15:
FYI... I removed the brackets around the -s, removed all spaces in the paths, and removed the brackets around the last path.  It tries to run but says it doesn't have permissions on my external drive.  When I check permissions from windows, it shows FULL ACCESS to everyone.  Not sure where to go from here.

Drewsif, 2017-07-15:
It looks like you have spaces in your path. You might try putting it in quotes like this `sudo /home/pi/Desktop/pishrink.sh "/media/pi/My Book/RetroPie Image/NoInputs128g_/NoInputs128g.img" "/media/pi/My Book/RetroPie Image/NoInputs128g_/NewNoInputs128g.img"` The brackets aren't required they just mean that you don't need to add that option