Drewsif/PiShrink #23: Systemd failed to start /etc/rc.local compatibility

j54n1n, 2017-10-05:
Hello,

I am getting an error while launching the /etc/rc.local file and thus also my partition does not get resized. It seems my rc.local file has trash in it.

I am [using your script](https://github.com/rainerum-robotics-rpi/raspbian-mods/blob/cd3fa8a9771e2900febce14f63519861af2e591c/prepare-rpi-image#L32) within a VirtualBox instance with Debian OS targeting a vbox shared folder of my host OS (Windows) containing a modified .img file cloned from a RPi SD card.

Maybe you have some insight on what is happening. Below you can find the related messages.

Output from serial console:
```
         Starting /etc/rc.local Compatibility...
         Starting Permit User Sessions...
[FAILED] Failed to start /etc/rc.local Compatibility.
See 'systemctl status rc-local.service' for details.
```
Output from `systemctl status rc-local.service`:
```
● rc-local.service - /etc/rc.local Compatibility
   Loaded: loaded (/lib/systemd/system/rc-local.service; static; vendor preset: enabled)
  Drop-In: /lib/systemd/system/rc-local.service.d
           └─debian.conf
        /etc/systemd/system/rc-local.service.d
           └─ttyoutput.conf
   Active: failed (Result: exit-code) since Sat 2017-09-30 10:36:21 UTC; 5min ago
  Process: 312 ExecStart=/etc/rc.local start (code=exited, status=203/EXEC)

Sep 30 10:36:20 raspberrypi systemd[1]: Starting /etc/rc.local Compatibility...
Sep 30 10:36:21 raspberrypi systemd[1]: rc-local.service: Control process exited, code=exited status=203
Sep 30 10:36:21 raspberrypi systemd[1]: Failed to start /etc/rc.local Compatibility.
Sep 30 10:36:21 raspberrypi systemd[1]: rc-local.service: Unit entered failed state.
Sep 30 10:36:21 raspberrypi systemd[1]: rc-local.service: Failed with result 'exit-code'.
Warning: rc-local.service changed on disk. Run 'systemctl daemon-reload' to reload units.
```
Contents of the file `/etc/rc.local` (seems to contain UTF-16 encoded stuff, so maybe filesystem related or vbox related!?):
```
PFKF@▒▒
▒ (F▒FF▒▒▒▒▒▒D▒F▒d▒▒▒0▒▒ %` q▒▒▒▒▒▒-▒▒▒-▒▒▒▒
▒▒P▒!PFnh@▒▒
```
Contents of the file `/etc/rc.local.bak`:
```
#!/bin/sh -e
#
# rc.local
#
# This script is executed at the end of each multiuser runlevel.
# Make sure that the script will "exit 0" on success or any other
# value on error.
#
# In order to enable or disable this script just change the execution
# bits.
#
# By default this script does nothing.

# Print the IP address
_IP=$(hostname -I) || true
if [ "$_IP" ]; then
  printf "My IP address is %s\n" "$_IP"
fi

exit 0
```
Thanks in advance,
Julian

j54n1n, 2017-10-05:
Hello again,

Now I have tried to run the script with all files within the VM and it works now witout flaws. It seems VirtualBox is screwing up something when the files are within the vbox shared folder.

So you can ignore my previous message.

Cheers Julian

closed 2017-10-05 by j54n1n

JavanXD, 2019-03-10:
@j54n1n  I have the same issue with my pi-shrinked image. Can you tell me how you fixed this issue?