Drewsif/PiShrink #24: Stretch lite image boot fails

broo0ose, 2017-10-23:
Hi,

on first boot I got 

`[      1.618965]    mmcblk0:  p1  p2 `

and then the Pi Zero froze. 

I've solved the problem so I though I'd post here to share the knowledge. 
After resizing an image the first boot fails due to (I think) the changed naming of the partitions in /etc/fstab and cmdline.txt. 

My solution after much googling was to replace:

 PARTUUID=0e75fcd5-01 with /dev/mmcblk0p1 
and
PARTUUID=0e75fcd5-02 with /dev/mmcblk0p2

in these files. 

I did it by mounting the partitions, but making the change before creating the full fat image would probably be easier. Unless it involves creating a time machine. 

HTH

Bruce

Drewsif, 2017-10-24:
I'm having trouble replicating your issue. Are you using the latest stretch image? And whats the OS you're running pishrink from?

broo0ose, 2017-10-24:
I'm running pishrink from Raspbian Stretch lite on a Pi3.  It was certainly downloaded as stretch and updated since, I don't know how many stretch Raspbian versions there have been so far.  

Not sure if this is relevant but the image I was shrinking was a Pi-Zero gadget image, so the cmdline.txt had been edited. The link I added in the first post pointed me to an issue with PARTUUID and sure enough explicitly putting the /dev entries in fixed the issue.

Drewsif, 2017-10-24:
Hmm, so the image I was using was booting with PARTUUID as well and I checked to make sure pisrink wasn't changing the UUID and it didn't for me. If it's the case that it is changing it I'd like to make it not do that.

I didn't run pishrink from a raspbian image, so let me try that and see if I can replicate it.

dazz100, 2018-06-05:
Hi
I am having a similar issue with piclone.  I run piclone on a headless Raspi with:
`PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
`
The cloned image won't boot.
I don't have a monitor connected so I can't tell you what messages are being outputted.  I can see from the flashing green LED that it is failing early in the boot process.  

piclone is not working on multiple pi's running different software.  I have tried different SD cards.  I have tried doing full (not just quick) SD formats. 

Let me know if I can provide additional info to help troubleshoot this problem.

Dazz

samvrlewis, 2019-01-09:
Also had this issue - the `PARTUUID`s definitely changed between the shrunk and non shrunk images. Maybe different versions of `parted` do this differently? For reference, I'm using Fedora 28 and parted 3.2.

The workaround I have been using is to use `blkid` to find the new `PARTUUID` of the partitions (on the shrunk image) and then updating `/etc/fstab` and `/boot/cmdline.txt` to refer to them.

Drewsif, 2023-02-14:
Closing for #255. Thank you for reporting the issue

closed 2023-02-14 by Drewsif