Drewsif/PiShrink #25: Error: Can't have overlapping partitions

latte47it, 2017-10-25:
Shrink doesn't complete correctly
command line:
sudo ./pishrink.sh  /media/sf_condivisione/tjbot_2.img immagine.jmg
Copying /media/sf_condivisione/tjbot_2.img to immagine.jmg...
Creating new /etc/rc.local
e2fsck 1.42.13 (17-May-2015)
Pass 1: Checking inodes, blocks, and sizes
Pass 2: Checking directory structure
Pass 3: Checking directory connectivity
Pass 4: Checking reference counts
Pass 5: Checking group summary information
root: 166841/863264 files (0.3% non-contiguous), 1371367/3451264 blocks
resize2fs 1.42.13 (17-May-2015)
resize2fs 1.42.13 (17-May-2015)
Resizing the filesystem on /dev/loop0 to 1701696 (4k) blocks.
Begin pass 2 (max = 405323)
Relocating blocks             XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
Begin pass 3 (max = 106)
Scanning inode table          XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
Begin pass 4 (max = 17858)
Updating inode references     XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
The filesystem on /dev/loop0 is now 1701696 (4k) blocks long.

Error: Can't have overlapping partitions.

--------
fdisk print of partition table of input image:

mauro@mauro-VirtualBox:~$ sudo fdisk /media/sf_condivisione/tjbot_2.img

Welcome to fdisk (util-linux 2.27.1).
Changes will remain in memory only, until you decide to write them.
Be careful before using the write command.

Command (m for help): p
Disk /media/sf_condivisione/tjbot_2.img: 14,9 GiB, 15931539456 bytes, 31116288 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x00052c15

Device                              Boot   Start      End  Sectors  Size Id Type
/media/sf_condivisione/tjbot_2.img1         2048  3287109  3285062  1,6G  e W95 FAT16 (LBA)
/media/sf_condivisione/tjbot_2.img2      3287110 31116287 27829178 13,3G  5 Extended
/media/sf_condivisione/tjbot_2.img5      3293184  3358717    65534   32M 83 Linux
/media/sf_condivisione/tjbot_2.img6      3358720  3500031   141312   69M  c W95 FAT32 (LBA)
/media/sf_condivisione/tjbot_2.img7      3506176 31116287 27610112 13,2G 83 Linux
----
any idea on how to overcame this problem?

Thanks in advance,
Mauro

Drewsif, 2017-10-26:
What kind of image is it? I've not run into this error before with any of the pi distros I've used.

latte47it, 2017-10-26:
Operating system of raspberry pi 3 is scratch
output of uname and /etc/os-release commands follows
let me know if you need other information
thanks for the assistance
Mauro
-----
pi@tjbot_mauro_2:~ $ uname -a
Linux tjbot_mauro_2 4.9.41-v7+ #1023 SMP Tue Aug 8 16:00:15 BST 2017 armv7l GNU/Linux
-----
pi@tjbot_mauro_2:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

Drewsif, 2017-12-02:
I think NOOBS does something weird to the partitions. I remember seeing someone post about it somewhere. I'll look into it more

lurch, 2018-01-29:
I've added some notes about this here: https://github.com/Drewsif/PiShrink/issues/44#issuecomment-361191005

Drewsif, 2018-02-03:
Closing this for #44 since it's a NOOBS issue

closed 2018-02-03 by Drewsif