Drewsif/PiShrink #32: Running pishrink via cron fails

K-Ko, 2017-12-19:
I run backups of my Raspberries onto a NAS.

These images should be shrinked nightly via cron, but

    Copying /mnt/backup/pi/pis0-20171031.img to /mnt/backup/pi/shrinked/pis0-20171031.img...
    Creating new /etc/rc.local
    e2fsck 1.43.4 (31-Jan-2017)
    e2fsck: A terminal is needed for interactive repairs
    resize2fs 1.43.4 (31-Jan-2017)
    resize2fs 1.43.4 (31-Jan-2017)
    Please run first "e2fsck -f / dev / loop0".
    
    ERROR: resize2fs failed...
OS

    Linux nas 3.16.0-4-amd64 #1 SMP Debian 3.16.43-2 (2017-04-30) x86_64 GNU/Linux

    PRETTY_NAME="Debian GNU/Linux 9 (stretch)"
    NAME="Debian GNU/Linux"
    VERSION_ID="9"
    VERSION="9 (stretch)"
    ID=debian
    HOME_URL="https://www.debian.org/"
    SUPPORT_URL="https://www.debian.org/support"
    BUG_REPORT_URL="https://bugs.debian.org/"
Is there chance to run it in background?

kmbach, 2017-12-29:
I just had this problem under Jessie right now. For me, e2fsck -p helped.

framps, 2017-12-29:
I create PR https://github.com/Drewsif/PiShrink/pull/34 which will fix the issue.

commit c40b4430 (2018-01-16): Merge pull request #34 from framps/master

Autorepair e2fsck errors
Closes #32
closed 2018-01-16 by Drewsif