Drewsif/PiShrink #35: PiShrink fails on latest Debian Stretch

DismissedGuy, 2018-01-02:
When running PiShrink on Debian 9, I get a lot of *2fs and mount errors:

$ sudo ./pishrink.sh ~/Desktop/shared/FULL_raspbian+RetroPie.img 
tune2fs: Input/output error while trying to open /dev/loop0
Couldn't find valid filesystem superblock.
tune2fs: Input/output error while trying to open /dev/loop0
Couldn't find valid filesystem superblock.
mount: /dev/loop0 is write-protected, mounting read-only
mount: wrong fs type, bad option, bad superblock on /dev/loop0,
       missing codepage or helper program, or other error
In some cases useful info is found in syslog - try
       dmesg | tail or so.
md5sum: /tmp/tmp.o0QqAkNH7P/etc/rc.local: No such file or directory
./pishrink.sh: line 63: [: !=: unary operator expected
umount: /tmp/tmp.o0QqAkNH7P: not mounted
e2fsck 1.43.4 (31-Jan-2017)
e2fsck: Operation not permitted while trying to open /dev/loop0
You must have r/w access to the filesystem or be root
resize2fs 1.43.4 (31-Jan-2017)
resize2fs: Operation not permitted while trying to open /dev/loop0
./pishrink.sh: line 136: [[: Couldn'tfindvalidfilesystemsuperblock.: syntax error: invalid arithmetic operator (error token is "'tfindvalidfilesystemsuperblock.")
expr: syntax error
expr: syntax error
expr: syntax error
resize2fs 1.43.4 (31-Jan-2017)
resize2fs: Operation not permitted while trying to open /dev/loop0
Couldn't find valid filesystem superblock.
ERROR: resize2fs failed...
mount: /dev/loop0 is write-protected, mounting read-only
mount: wrong fs type, bad option, bad superblock on /dev/loop0,
       missing codepage or helper program, or other error
In some cases useful info is found in syslog - try
       dmesg | tail or so.
mv: cannot stat '/tmp/tmp.o0QqAkNH7P/etc/rc.local.bak': No such file or directory
umount: /tmp/tmp.o0QqAkNH7P: not mounted`

$ sudo dmesg | tail
[ 1636.561900] UDF-fs: error (device loop0): udf_read_tagged: read failed, block=5911639, location=5911639
[ 1636.561904] UDF-fs: error (device loop0): udf_read_tagged: read failed, block=5911747, location=5911747
[ 1636.561909] UDF-fs: error (device loop0): udf_read_tagged: read failed, block=5911491, location=5911491
[ 1636.561913] UDF-fs: error (device loop0): udf_read_tagged: read failed, block=5911745, location=5911745
[ 1636.561917] UDF-fs: error (device loop0): udf_read_tagged: read failed, block=5911489, location=5911489
[ 1636.561921] UDF-fs: error (device loop0): udf_read_tagged: read failed, block=512, location=512
[ 1636.561925] UDF-fs: error (device loop0): udf_read_tagged: read failed, block=256, location=256
[ 1636.561928] UDF-fs: error (device loop0): udf_read_tagged: read failed, block=512, location=512
[ 1636.561930] UDF-fs: warning (device loop0): udf_fill_super: No partition found (1)
[ 1636.571291] isofs_fill_super: bread failed, dev=loop0, iso_blknum=16, block=32

Can anybody tell me what the hell is going on? The errors seem to have to do with no permissions to me, but I'm sure I'm running the script as root.
P.S.: Please don't look at my formatting :)

Drewsif, 2018-01-16:
How did you take the image? It looks like that might be broken

vdonchev, 2018-01-18:
Have the same problem on latest mint64:

tune2fs: Attempt to read block from filesystem resulted in short read while trying to open /dev/loop0
Couldn't find valid filesystem superblock.
tune2fs: Attempt to read block from filesystem resulted in short read while trying to open /dev/loop0
Couldn't find valid filesystem superblock.
mount: /dev/loop0 is write-protected, mounting read-only
mount: /dev/loop0: can't read superblock
md5sum: /tmp/tmp.AmJQUNDFi9/etc/rc.local: No such file or directory
./pishrink.sh: line 63: [: !=: unary operator expected
umount: /tmp/tmp.AmJQUNDFi9: not mounted
e2fsck: Operation not permitted while trying to open /dev/loop0
You must have r/w access to the filesystem or be root
resize2fs 1.42.13 (17-May-2015)
resize2fs: Operation not permitted while trying to open /dev/loop0
./pishrink.sh: line 136: [[: Couldn'tfindvalidfilesystemsuperblock.: syntax error: invalid arithmetic operator (error token is "'tfindvalidfilesystemsuperblock.")
expr: syntax error
expr: syntax error
expr: syntax error
resize2fs 1.42.13 (17-May-2015)
resize2fs: Operation not permitted while trying to open /dev/loop0
Couldn't find valid filesystem superblock.
ERROR: resize2fs failed...
mount: /dev/loop0 is write-protected, mounting read-only
mount: /dev/loop0: can't read superblock
mv: cannot stat '/tmp/tmp.AmJQUNDFi9/etc/rc.local.bak': No such file or directory
umount: /tmp/tmp.AmJQUNDFi9: not mounted

any ideas?

DismissedGuy, 2018-01-22:
By the way: one of the images is a NOOBS dual boot, but since the other one doesn't work either, I don't think that's the problem.

John-Mc, 2018-01-27:
Working fine on the latest Raspian-stretch.

Drewsif, 2018-02-03:
NOOBS is a known problem. They do partitioning different than raspbian and it breaks the script

DismissedGuy, 2018-02-04:
But that shouldn't break the other image, should it?

Drewsif, 2018-02-04:
It shouldn't. Have you tried that latest version to see if anything got fixed?

You're running PiShrink on a standard Debian 9 install? I'll try to do it as well and see if I can replicate the error.

himijendrix24, 2018-02-16:
Same here: `sudo pishrink.sh Raspi3_basic.img`

Debian Stretch 64 bit (virtualbox with windows 10 host).
There is 35 Gb free disk space for the virtual machine and
the image is accessed via a shared folder on the host system. 

_tune2fs: Eingabe-/Ausgabefehler beim Versuch, /dev/loop0 zu öffnen
Es kann kein gültiger Dateisystem-Superblock gefunden werden.
mount: /dev/loop0 is write-protected, mounting read-only
mount: wrong fs type, bad option, bad superblock on /dev/loop0,
       missing codepage or helper program, or other error
       In some cases useful info is found in syslog - try
       dmesg | tail or so.
md5sum: /tmp/tmp.lMI5EW1kNW/etc/rc.local: Datei oder Verzeichnis nicht gefunden
/usr/local/bin/pishrink.sh: Zeile 69: [: !=: Einstelliger (unärer) Operator erwartet.
umount: /tmp/tmp.lMI5EW1kNW: not mounted
e2fsck: Die Operation ist nicht erlaubt beim Versuch, /dev/loop0 zu öffnen
Sie benötigen r/w- oder root-Rechte für das Dateisystem.
resize2fs 1.43.4 (31-Jan-2017)
resize2fs: Die Operation ist nicht erlaubt beim Versuch, /dev/loop0 zu öffnen
/usr/local/bin/pishrink.sh: Zeile 142: [[: EskannkeingültigerDateisystem-Superblockgefundenwerden.: Syntaxfehler: Ungültiger arithmetischer Operator. (Fehlerverursachendes Zeichen ist \"ültigerDateisystem-Superblockgefundenwerden.\").
/usr/local/bin/pishrink.sh: Zeile 148: - EskannkeingültigerDateisystem-Superblockgefundenwerden.: Syntaxfehler: Ungültiger arithmetischer Operator. (Fehlerverursachendes Zeichen ist \"ültigerDateisystem-Superblockgefundenwerden.\").
resize2fs 1.43.4 (31-Jan-2017)
resize2fs: Die Operation ist nicht erlaubt beim Versuch, /dev/loop0 zu öffnen
Es kann kein gültiger Dateisystem-Superblock gefunden werden.
ERROR: resize2fs failed...
mount: /dev/loop0 is write-protected, mounting read-only
mount: wrong fs type, bad option, bad superblock on /dev/loop0,
       missing codepage or helper program, or other error In some cases useful info is found in syslog - try dmesg | tail or so.
mv: der Aufruf von stat für '/tmp/tmp.lMI5EW1kNW/etc/rc.local.bak' ist nicht möglich: Datei oder Verzeichnis nicht gefunden
umount: /tmp/tmp.lMI5EW1kNW: not mounted_

molehole2003, 2018-03-15:
Using Mint 18.3 (Virtualbox with windows 10 host) and the image is accessed via a shared folder on the host system. There appears to be an issue with running PiShrink and the newest version 5.2.8 of Virtualbox Guest additions.  When I installed Mint 18.1 and left the pre-installed Guest additions, PiShrink works fine. When I updated VB Guest additions in Mint 18.1 to version 5.2.8, I get the same error messages I was getting in Mint 18.3.

tune2fs: Attempt to read block from filesystem resulted in short read while trying to open /dev/loop0
Couldn't find valid filesystem superblock.
mount: /dev/loop0 is write-protected, mounting read-only
mount: /dev/loop0: can't read superblock
md5sum: /tmp/tmp.IVWSnYvrca/etc/rc.local: No such file or directory
./pishrink.sh: line 69: [: !=: unary operator expected
umount: /tmp/tmp.IVWSnYvrca: not mounted
e2fsck: Operation not permitted while trying to open /dev/loop0
You must have r/w access to the filesystem or be root
resize2fs 1.42.13 (17-May-2015)
resize2fs: Operation not permitted while trying to open /dev/loop0
./pishrink.sh: line 142: [[: Couldn'tfindvalidfilesystemsuperblock.: syntax error: invalid arithmetic operator (error token is "'tfindvalidfilesystemsuperblock.")
./pishrink.sh: line 148: - Couldn'tfindvalidfilesystemsuperblock.: syntax error: invalid arithmetic operator (error token is "'tfindvalidfilesystemsuperblock.")
resize2fs 1.42.13 (17-May-2015)
resize2fs: Operation not permitted while trying to open /dev/loop0
Couldn't find valid filesystem superblock.
ERROR: resize2fs failed
...

I am no Linux or Virtualbox expert, but this seems like another clue to some of the problems listed above.

steeeb, 2018-03-19:
Same happened here, I downloaded the 5.2 guest additions and installed them instead and it fixed the issue.

aitgelion, 2018-03-21:
I have the same problem with ubuntu 16.04.3 (32b) running on Virtualbox (Win10).
I solved it copying the .img file from shared Virtualbox Folder to Virtual Machine home directory and running the script from there.

PatiB87, 2018-08-16:
Hey, I got the same issues. I tried Linux MINT and Debian on Windows 10 Host and MacOS Host, I tried having the image on an external HDD connected to the VM and also tried to copy the image to the virtual disk of the VM. It all resulted in the same errors as described above. "Bad superblock" etc...
What is going on? Is PiShrink broken by now?
I used it successfully about a year ago, what changed?

bjcerveny, 2018-08-20:
@PatiB87 I just performed a pishrink today in Ubuntu 18.04 on my Mac. It worked fine after copying the image to the virtual disk.

PatiB87, 2018-08-20:
How did you create the image?
Did you use 32 bit Ubuntu or 64?

bjcerveny, 2018-08-20:
I made two images while debugging. One using 'dd' on my Mac and one directly from Ubuntu. They both work when copied to the VM's filesystem. Neither work when accessed via a share on the host Mac.

PatiB87, 2018-08-20:
Interesting! Did you use the /rdisk or the /disk path with dd on the mac? Did you modify the block size as well?
But your Ubuntu is 64 bit? And english? (I somehow suspect that maybe my German installation might be the reason why...)

PatiB87, 2018-09-15:
Tried it exactly as you said. Same command under Mac, image making went fine, copied it to my freshly installed Ubuntu 64-bit English in VirtualBox, ran latest pishrink and got this:
[https://pictr.com/image/0aCAQ1](url)

All this for nothing.

Could it be that your Ubuntu is still old and there IS in fact a problem with PiShrink on newer Linux version as suggested by so many bug reports and forum entries?

bjcerveny, 2018-09-16:
It is a fresh install of 18.04. I do not have another version I will be making a new image later this week.
I do use Parallels, not VirtualBox.

ab316, 2018-10-14:
Worked from me after installing Virtual Box Guest Additions 5.2.0 and moving the image from the shared folder to inside the machine.

Websteria, 2019-01-09:
This needs to be fixed. Pishrink is basically broken for any x86 users with the current version. I’ve tried everything and can’t get it to work unless I copy the file into the target machine.

rmast, 2019-01-23:
I just ran this fork-commit of pishrink.sh without errors on my RPi3-image, hopefully it will grow again when I need it:
https://raw.githubusercontent.com/simonj0/PiShrink/2b4a63299ef25b20142122a2fec9e3209fcd5380/pishrink.sh

MapledLeif, 2019-03-13:
I was getting the "Couldn't find valid filesystem superblock" error on virtual machines running Mint and Ubuntu with the host on my Win10 as well as Mac Mojave utilizing shared folders.  Only when copying into the virtual machine would it work.

Finally I just connected a large enough HDD via powered USB hub to my RPi running Raspbian Stretch and it worked like a charm the first try.  No fuss, no muss, and shrunk a 128GB card from 127.84 to just over 109 in about 10 minutes.

sshanky, 2019-07-08:
I was having trouble using Ubuntu 16.04 in Parallels on my Mac via a shared filesystem. Once I copied the image file off the Mac's filesystem onto the local Ubuntu filesystem, it worked perfectly, going from nearly 8GB down to 3.7. Awesome. Thanks everyone.

Duiesel, 2020-02-16:
I've got the same problem on my mac os and pishrink in Docker.
```chmod -R 777 my_image_file.img``` before ```docker run``` helps.
Good luck!

galosre, 2020-06-29:
Me i am getting:!!
 Tune2fs: Bad magic number in super-block while trying to open /dev/loop0
/dev/loop0 contain a swap file system
pishrink.sh: ERROR occurred in line 289: tune2fs failed. Unable to shrink this type of image

Drewsif, 2020-07-01:
@galosre your image is not shrinkable with PiShrink. The last partition needs to be the root partition and yours is the swap partition. What OS are you using?

sookah, 2020-07-10:
> I've got the same problem on my Mac os and pishrink in Docker (Debian).
> `chmod -R 777 my_image_file.img` before `docker run` and **"pishink'ing"** helps.
> The reason is that when you made `pishring file.img` you trying to rewrite files in mounted image file so **pishrink** can't do so on read-only mount.
> Good luck!

@Duiesel I am doing this exactly right now on docker ubuntu on MacOS. What do you mean by before docker run, care to explain it a bit better would really help me right now. 

What i did was `docker cp <file> <destination>`. to copy the image into the container and than run pi shrink there. Failed with the above error.

kentsimp, 2020-07-19:
> I've got the same problem on my Mac os and pishrink in Docker (Debian).
> `chmod -R 777 my_image_file.img` before `docker run` and **"pishink'ing"** helps.
> The reason is that when you made `pishring [...]

This put me on the right track. Using windows though, I just changed the output file so I don't need write permissions:

`./pishrink.sh input-image.img output-image.img`

mgomesborges, 2021-02-10:
Hey @DismissedGuy @vdonchev @Duiesel @sookah you can check out my project [PiShrink dockerized](https://github.com/mgomesborges/pishrink-docker) | [Docker Hub PiShrink](https://hub.docker.com/r/mgomesborges/pishrink):

You can see how I create the Ubuntu Docker image, the installation process is very similar for Debian.

## macOS setup

1. [Install Docker Desktop on Mac](https://docs.docker.com/docker-for-mac/install/)

2. Make a copy of a Raspberry Pi SD card that you want to shrink ([see instructions here](https://github.com/mgomesborges/raspberry-pi/blob/master/setup/clone-sd-card.md)).

3. Using the Terminal, access the directory containing the Raspberry Pi image:

```bash
cd ~/Directory-with-RPi-image
```

4. Run PiShrink dockerized:

```bash
docker run --privileged=true --rm \
    --volume $(pwd):/workdir \
    mgomesborges/pishrink \
    pishrink -pZv IMAGE_NAME.img NEW-IMAGE_NAME.img
```

Pomurphy, 2021-08-18:
> I just ran this fork-commit of pishrink.sh without errors on my RPi3-image, hopefully it will grow again when I need it:
> https://raw.githubusercontent.com/simonj0/PiShrink/2b4a63299ef25b20142122a2fec9e3209fcd5380/pishrink.sh

This helped me to transform a Kali 64 bits image of 64Gb to a 20Gb img file

dogarcar, 2022-03-02:
> @galosre your image is not shrinkable with PiShrink. The last partition needs to be the root partition and yours is the swap partition. What distro are you using on the pi?

I have indeed the same issue. I'm using PiShrink to fit a backup sd card image into a new (slightly smaller) sd card. It's a customized RetrOZ image. I reduced the size of the image (the rom partition) as it was plenty of space not used and now I'm trying to shrink it, but it does not work.

I'm using the ubuntu system (version 2) installed in windows 10.

Any help to make it work? Thanks!!!

AndreCox, 2022-05-13:
Has anyone tried `swapoff -a` then shutting down their raspberry pi, I'm testing it now to see if it works.

AndreCox, 2022-05-13:
Can confirm that doesn't work

closed 2023-02-07 by Drewsif