Drewsif/PiShrink #41: Odd progress report from Pishrink?

John-Mc, 2018-01-27:
After shrinking a couple of images successfully (and getting progress reports similar to what is shown on the "Example" on your code page), I'm not getting an abbreviated printout:
`Copying /mnt/pi-usbdrive/raspi-stretch2.img to /mnt/pi-usbdrive/raspi-str2-sml.img...
Creating new /etc/rc.local
rootfs: 137443/957712 files (0.2% non-contiguous), 1150083/3877760 blocks
resize2fs 1.43.4 (31-Jan-2017)
resize2fs 1.43.4 (31-Jan-2017)
Resizing the filesystem on /dev/loop0 to 1574368 (4k) blocks.
Begin pass 3 (max = 119)
Scanning inode table          XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
The filesystem on /dev/loop0 is now 1574368 (4k) blocks long.

Shrunk /mnt/pi-usbdrive/raspi-str2-sml.img from 15G to 6.1G`

No error reports, but seems odd that I am not getting the rest of the status report  (pass 1, pass 2, pass 3... etc.)

Is this anything to worry about

lurch, 2018-01-29:
I believe it's entirely up to `resize2fs` what progress reports get printed, so as long as there are no errors it's probably nothing to worry about?
https://linux.die.net/man/8/resize2fs

John-Mc, 2018-01-29:
Thanks @lurch.  No errors reported. I just thought it was odd, since it used to report the whole sequence when I ran it.  Could the difference be that the first few times I ran it, the image was on an SD card, but more recently it was on an ExFAT formatted thumb drive (I had installed exfat-fuse and exfat-utils﻿ so my R-Pi3 could deal with this.)

lurch, 2018-01-29:
I've got absolutely no idea... my "best guess" would be that maybe `resize2fs` is able to skip certain steps if it can detect that they've already been run or don't need to be run? \*shrug\*

Drewsif, 2018-02-03:
Its hard for me to predict why it would be giving you different output. I wouldn't worry about it since it looks like everything worked!

closed 2018-02-03 by Drewsif

John-Mc, 2018-02-03:
Thanks. I've checked and the images I shrink do work. I suspect @lurch may be onto something. I was shrinking brand new images: installed an image, did apt-get update/upgrade, made some minimal edits and then shrunk it. I guess not enough messed up on it yet to require some of the other scans to run.