Drewsif/PiShrink #42: "sudo ./pishrink.sh" vs "sudo pishrink.sh" ??

John-Mc, 2018-01-27:
I can never get PiShrink to run using the command in your example:
`sudo ./pishrink.sh ...`
If I omit the leading ./ everything runs fine.

Did I make an error in my installation? (I followed the commands suggested by @framps in issue #36.)

Just wondering if I did something wrong, or if the ./ should be deleted from your example.

framps, 2018-01-27:
I updated the README and created PR https://github.com/Drewsif/PiShrink/pull/43

lurch, 2018-01-29:
FYI, running `sudo ./pishrink.sh ...` works if you _don't_ do `sudo mv pishrink.sh /usr/local/bin` ;-)  (but then you end up with the `pishrink.sh` 'hanging around' in your home directory).

More details, for reference:
You're correct, `./` does mean use a relative directory reference, it means look in the current directory. But since you've now moved `pishrink.sh` from the current directory to `/usr/local/bin`, you either need to use `sudo /usr/local/bin/pishrink.sh ...`, or you can just use `sudo pishrink.sh ...` since `/usr/local/bin` is included in the `$PATH` variable. When you run a 'bare command' with no path prefix, the shell (in your case bash) looks for the command in each of the directories specified in `$PATH`.

lurch, 2018-01-29:
@John-Mc You're close... :wink:  by default on Linux the current directory is never in the `$PATH` so if you have `pishrink.sh` in e.g. your home directory, then trying to run `sudo pishrink.sh ...` would fail, you either need to run `sudo ./pishrink.sh ...` or e.g. `sudo /home/pi/pishrink.sh ...`
If you give an explicit path to an executable (and `./` is an explicit path to the current directory), then the shell doesn't need to try searching the `$PATH` to find it :)

lurch, 2018-02-03:
@John-Mc Can this be closed, now that the README has been updated?

Drewsif, 2018-02-03:
Closing since its already merged

closed 2018-02-03 by Drewsif

John-Mc, 2018-02-03:
Sure. Sorry, I didn't know the etiquette of closing an issue. I figured it was up to the site owner.