Drewsif/PiShrink #44: NOOBS Pre-1.5 Partitions Issue

bmb11107, 2018-01-28:
The error I got running pishrink.sh was

`md5sum: /tmp/tmp.3KNSjKk0bS/etc/rc.local: No such file or directory`

This led me to discover that pishrink.sh is looking for the last entry "`tail -n 1`" in the partition table. In my case, with a pre-1.5 NOOBS install, this found the NOOBS partition (3) instead of the root partition of Raspbian (6). See [NOOBS Partitioning Explained](https://github.com/raspberrypi/noobs/wiki/NOOBS-partitioning-explained) Pishrink continued running, but at best the wrong partition was shrunk. More likely it didn't work at all. 

I tried adding support for a -p flag ("use_this_partition"), so I could specify "`-p 6`" and it would use `grep "^6:"` instead of `tail` which seemed to work, but then produced this error

`Error: Can't have overlapping partitions.`

which leads me to believe I am in over my head. Maybe pishrink doesn't work unless it's working on the *last* partition, or maybe it doesn't work on non-primary logical partitions, or maybe I just did something wrong to mess up the math. In any case, I think it would be cool if this could be fixed, but I think the right answer for me is to use a newer NOOBS, or more likely get rid of NOOBS and just run Raspbian as described [here](https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=62253)

Thanks for your attention.

lurch, 2018-01-29:
See also #25 (which is also concerned with NOOBS)

Drewsif, 2020-06-17:
Closing as a duplicate

closed 2020-06-17 by Drewsif