Drewsif/PiShrink #47: PiShrink using bootable SD card as source?

John-Mc, 2018-01-29:
Is it possible to use PiShrink using a bootable SD card as the source, or do I have to start from an .img file? 

I've been using the commands
`sudo dd if=/dev/mmcblk0 of=<pathname>raspi.img
sudo pishrink.sh <pathname>/raspi.img`
to copy the image to a USB flashdrive connected to my R-Pi first, then running PiShrink against the result. 

Is it possible to do 
`pishrink.sh /dev/mmcblk0 <pathname>raspi.img`
and eliminate an extra step?

framps, 2018-01-29:
From a scripting point of view it was possible until https://github.com/Drewsif/PiShrink/commit/2229ae89a642a90e14c5d25608c5edbf653c3f84 was merged. Why don't you just revert the change and give it a try. Don't forget to keep us updated :-)

John-Mc, 2018-01-29:
Undid the change from that merge, and it worked fine. However, it does not end up saving time. I had forgotten that when PiShrink gives a new filename, it copies the file first, then works on the copied file to shrink that. It saves me typing in 1 command, but that is not such a big deal. It's still going through the same steps and taking the same amount of Pi time as it would with me entering the two commands. So I changed it back to the original.

closed 2018-01-29 by John-Mc