Drewsif/PiShrink #52: **_md5sum: /tmp/tmp.kpSNXt6TfD/etc/rc.local: No such file or directory_**

brainstewx, 2018-02-14:
I appreciate all that you do. I just haven't been able to get PiShrink to work. This is a log of my latest attempt to run PiShrink. My computer is a Raspberry Pi 3 running the latest version of Raspbian. PiShrink.sh and the recalbox410.img are on my external hard drive. The resulting image is unusable. It seems a temp folder with rc.local missing. I can see in my File Manager that the folder is created. I also tried this from Ubuntu MATE and received the same error. My Recalbox image has about 3 gigabytes of free space. 

Just FYI. Thanks in advance, if you do read this. 

pi@raspberrypi:/media/pi/FreeAgentDrive/PiShrink-master $ sudo ./pishrink.sh recalbox410.img
**_md5sum: /tmp/tmp.kpSNXt6TfD/etc/rc.local: No such file or directory_**
./pishrink.sh: line 69: [: !=: unary operator expected
share0: 8149/794624 files (0.7% non-contiguous), 2407673/3169792 blocks
resize2fs 1.43.4 (31-Jan-2017)
resize2fs 1.43.4 (31-Jan-2017)
Resizing the filesystem on /dev/loop0 to 2883672 (4k) blocks.
Begin pass 2 (max = 216118)
Relocating blocks             XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
Begin pass 3 (max = 97)
Scanning inode table          XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
The filesystem on /dev/loop0 is now 2883584 (4k) blocks long.

Error: You requested a partition from 2822766592B to 14634287104B (sectors 5513216..28582592).
The closest location we can manage is 15816720384B to 15816720384B (sectors 30892032..30892032).
Shrunk recalbox410.img from 15G to 15G

Drewsif, 2020-06-17:
Should be fixed in the new release!

closed 2020-06-17 by Drewsif

cpdk, 2020-06-17:
Can confirm this is now working - was googling this and found this issue only to see it just closed. Only comment is that version number at top of script does not seem to have changed. 

But great work with this - thanks a lot.

Drewsif, 2020-06-17:
One day I'll remember to bump the version number... One day...

framps, 2020-06-17:
Keep in mind that's the only way pishrink users are able to check whether there is a new release available. I don't think most of the will diff their local version with the current git release :wink: 

You can add code to pishrink to check whether there is a new release and notify the user in a message of the new version available. That way they will not use old releases ... JM2C