Drewsif/PiShrink #54: Loop device remains open on error and abort

kmbach, 2018-02-14:
If pishrink does not leave properly, the loop device remains open. Memory is blocked until the loop device is deleted. A trap handler could solve this Problem.

function cleanup() {
    losetup -d "$loopback"
}

trap cleanup SIGINT SIGTERM EXIT 

Please check and implement if necessary.

framps, 2018-02-14:
@kmbach I like your proposal and modified pishrink in [url=https://github.com/framps/PiShrink/blob/master/pishrink.sh]my fork[/url]. Please test whether it's working as expected. If yes, let me know and I will create a PR.

kmbach, 2018-02-15:
Hi, framp, I provokad a 'ERROR: Image already shrunk to smalllest size'
And as expected - the loop drive was deleted.

commit fd6c0b1b (2018-03-30): Merge pull request #55 from framps/master

Added cleanup of loop device in case of execution errors
Closes #54
closed 2018-03-30 by Drewsif