Drewsif/PiShrink #56: fail tu run skript

brainnex, 2018-02-19:
Hello,
sorry if this is a noob question but I can't get it working. I'm running  raspbian jessie and trying to resize an image from a windows network share. This is all what i get

```
 sudo ~/bin/pishrink.sh /mnt/network/1.img
Creating new /etc/rc.local
/dev/loop0: Inodes wurden gefunden, die Teil einer defekten verketteten Liste von
verwaisten Inodes waren.

/dev/loop0: UNERWARTETE INKONSISTENZ; fsck MANUELL AUSFÜHREN
        (d.h. ohne die Optionen -a oder -p)
resize2fs 1.43.3 (04-Sep-2016)
Bitte lassen Sie zuerst „e2fsck -f /dev/loop0“ laufen.

/home/pi/bin/pishrink.sh: Zeile 148: 7659520 - : Syntax Fehler: Operator erwartet. (Fehlerverursachendes Zeichen ist \"- \").
resize2fs 1.43.3 (04-Sep-2016)
Bitte lassen Sie zuerst „e2fsck -f /dev/loop0“ laufen.

ERROR: resize2fs failed...

```

framps, 2018-02-20:
@brainnex I suggest to call pishrink.sh again the following way:
`sudo LANG="en_US.UTF-8"  /home/pi/bin/pishrink.sh /mnt/network/1.img`
and attach the output. That way all messages will be in English and everybody can read them ;-)

Backolous, 2018-03-30:
It works! Thank you!

You have to execute the script twice. It repairs  the File in the first run, but fails to finish. If you start again it works. The created Image is fine, booting without problems. I was able to replace a 32 gb card with a 2 gb one.
Here is the output:
```

user@UBUNTU:~/pishrink$ sudo ./pishrink.sh -s raspberry-backup.img
[sudo] password for luser: 
Skipping autoexpanding process...
/dev/loop0: Inodes that were part of a corrupted orphan linked list found.  

/dev/loop0: UNEXPECTED INCONSISTENCY; RUN fsck MANUALLY.
	(i.e., without -a or -p options)
Error 4 occured when checking filesystem. Retrying ...
e2fsck 1.43.5 (04-Aug-2017)
Pass 1: Checking inodes, blocks, and sizes
Inodes that were part of a corrupted orphan linked list found.  Fix? yes

Inode 1507411 was part of the orphaned inode list.  FIXED.
Inode 1638572 was part of the orphaned inode list.  FIXED.
Pass 2: Checking directory structure
Pass 3: Checking directory connectivity
Pass 4: Checking reference counts
Pass 5: Checking group summary information

/dev/loop0: ***** FILE SYSTEM WAS MODIFIED *****
/dev/loop0: 2860/1792000 files (1.1% non-contiguous), 322388/7168000 blocks
resize2fs 1.43.5 (04-Aug-2017)
resize2fs: Invalid argument while trying to open /dev/loop0
./pishrink.sh: line 162: [[: Couldn'tfindvalidfilesystemsuperblock.: syntax error: invalid arithmetic operator (error token is "'tfindvalidfilesystemsuperblock.")
./pishrink.sh: line 168: 7168000 - Couldn'tfindvalidfilesystemsuperblock.: syntax error: invalid arithmetic operator (error token is "'tfindvalidfilesystemsuperblock.")
resize2fs 1.43.5 (04-Aug-2017)
resize2fs: Invalid argument while trying to open /dev/loop0
Couldn't find valid filesystem superblock.
ERROR: resize2fs failed...
mount: : mount point does not exist.
mv: cannot stat '/etc/rc.local.bak': No such file or directory
umount: : no mount point specified.
losetup: /dev/loop0: detach failed: No such device or address

user@UBUNTU:~/pishrink$ sudo ./pishrink.sh -s raspberry-backup.img
Skipping autoexpanding process...
/dev/loop0: 2860/1792000 files (1.1% non-contiguous), 322388/7168000 blocks
resize2fs 1.43.5 (04-Aug-2017)
resize2fs 1.43.5 (04-Aug-2017)
Resizing the filesystem on /dev/loop0 to 190160 (1k) blocks.
Begin pass 2 (max = 75090)
Relocating blocks             XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
Begin pass 3 (max = 875)
Scanning inode table          XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
Begin pass 4 (max = 593)
Updating inode references     XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
The filesystem on /dev/loop0 is now 190160 (1k) blocks long.

Shrunk raspberry-backup.img from 30G to 700M

```

framps, 2018-03-30:
Glad my fix helped. 

Actually you just have to execute  `e2fsck -f -y "$loopback"` to fix the image inconsistency and then execute pishrink again. I added this logic in pishrink but it seems the image has to be remounted first before shrinking the image :-(

brainnex, 2018-03-31:
I think my initial problem was related to the r/w rights of the windows network share. so maybe not a problem of the script itself. I didn't tried it again because i resized manually the image under ubuntu

himijendrix24, 2018-05-24:
I previously could also not run the script in a virtual machine and now tried on a ubuntu notebook. 
file r/w permissions are set, but I still get the following error:

`je@je-ThinkPad-X250:~/Öffentlich$ sudo LANG="en_US.UTF-8" pishrink.sh ./Raspi3_basic.img
/dev/loop18: Inodes that were part of a corrupted orphan linked list found.  

/dev/loop18: UNEXPECTED INCONSISTENCY; RUN fsck MANUALLY.
	(i.e., without -a or -p options)
resize2fs 1.44.1 (24-Mar-2018)
resize2fs: Invalid argument while trying to open /dev/loop18
/usr/local/bin/pishrink.sh: line 151: [[: Couldn'tfindvalidfilesystemsuperblock.: syntax error: invalid arithmetic operator (error token is "'tfindvalidfilesystemsuperblock.")
/usr/local/bin/pishrink.sh: line 157: 3769856 - Couldn'tfindvalidfilesystemsuperblock.: syntax error: invalid arithmetic operator (error token is "'tfindvalidfilesystemsuperblock.")
resize2fs 1.44.1 (24-Mar-2018)
resize2fs: Invalid argument while trying to open /dev/loop18
Couldn't find valid filesystem superblock.
ERROR: resize2fs failed...
mount: /tmp/tmp.6Pt1VmHGSF: can't read superblock on /dev/loop18.
mv: cannot stat '/tmp/tmp.6Pt1VmHGSF/etc/rc.local.bak': No such file or directory
umount: /tmp/tmp.6Pt1VmHGSF: not mounted.
losetup: /dev/loop18: detach failed: No such device or address`

gmenezesg, 2018-06-13:
I've had a similar issue. I've created a image file from a working Raspbian SDCard using Win32DiskImager. When I ran pishrink, the tool gave me the "orphaned inode list" error. So I followed the suggestion from Rocky Inde and executed fsck. It encountered and fixed some errors so I ran pishrink again and it worked! Thanks Rock Inde.

If you got this far and are still confused how to do it, I've created a script, partly based on pishrink, to fix these "orphaned inodes". You can check the script source at 

https://github.com/gmenezesg/fix_orphaned_inode_list

Hope it helps!

framps, 2018-06-13:
Thank you very much for sharing your code :+1: . 

As far as I can see you execute a fsck on the .img file in your code. Why didn't you just add this check in pishrink and create a PR  :smile: ?  I think it makes sense to execute fsck all the time before executing the resizing. But this will add some additional processing time. For people concerned about execution time an option to exclude fsck might be useful.

gmenezesg, 2018-06-13:
@framps That makes total sense. I will try to do it. Create a pull request and implement the fsck at the early steps of the script, triggered by a parameter, since it won't hurt if the fsck encounters nothing to fix. If so, it fixes and continues the shrinking process. Thanks for the tip mate!

sujitrp, 2018-07-23:
Hi Friends 
I am getting following error . Kindly suggest solution . 

sudo pishrink.sh /home/pi/mynetworkdrive/donottouch/raspberrypi_backups/PI_Backup/backup_raspberrypi_20171228_001853.img
Creating new /etc/rc.local
root0: 163864/873120 files (0.5% non-contiguous), 1341398/3489152 blocks
resize2fs 1.43.3 (04-Sep-2016)
resize2fs 1.43.3 (04-Sep-2016)
Resizing the filesystem on /dev/loop2 to 1671247 (4k) blocks.
Begin pass 2 (max = 501469)
Relocating blocks             XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
Begin pass 3 (max = 107)
Scanning inode table          XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
Begin pass 4 (max = 23511)
Updating inode references     XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
The filesystem on /dev/loop2 is now 1671168 (4k) blocks long.

Error: You requested a partition from 1639972864B to 8485400576B (sectors 3203072..16573048).
The closest location we can manage is 15931539456B to 15931539456B (sectors 31116288..31116288).
Shrunk /home/pi/mynetworkdrive/donottouch/raspberrypi_backups/PI_Backup/backup_raspberrypi_20171228_001853.img from 15G to 15G

sujitrp, 2018-07-23:
on 2nd try I am getting following error 
sudo pishrink.sh /home/pi/mynetworkdrive/donottouch/raspberrypi_backups/PI_Backup/backup_raspberrypi_20171228_001853.img
tune2fs: Bad magic number in super-block while trying to open /dev/loop2
mount: /dev/loop2: can't read superblock
md5sum: /tmp/tmp.BIPc4NBXGV/etc/rc.local: No such file or directory
/usr/local/bin/pishrink.sh: line 78: [: !=: unary operator expected
umount: /tmp/tmp.BIPc4NBXGV: not mounted
e2fsck: Invalid argument while trying to open /dev/loop2
/dev/loop2:
The superblock could not be read or does not describe a valid ext2/ext3/ext4
filesystem.  If the device is valid and it really contains an ext2/ext3/ext4
filesystem (and not swap or ufs or something else), then the superblock
is corrupt, and you might try running e2fsck with an alternate superblock:
    e2fsck -b 8193 <device>
 or
    e2fsck -b 32768 <device>

resize2fs 1.43.3 (04-Sep-2016)
resize2fs: Invalid argument while trying to open /dev/loop2
/usr/local/bin/pishrink.sh: line 151: [[: Couldn'tfindvalidfilesystemsuperblock.: syntax error: invalid arithmetic operator (error token is "'tfindvalidfilesystemsuperblock.")
/usr/local/bin/pishrink.sh: line 157: - Couldn'tfindvalidfilesystemsuperblock.: syntax error: invalid arithmetic operator (error token is "'tfindvalidfilesystemsuperblock.")
resize2fs 1.43.3 (04-Sep-2016)
resize2fs: Invalid argument while trying to open /dev/loop2
Couldn't find valid filesystem superblock.
ERROR: resize2fs failed...
mount: /dev/loop2: can't read superblock
mv: cannot stat ‘/tmp/tmp.BIPc4NBXGV/etc/rc.local.bak’: No such file or directory
umount: /tmp/tmp.BIPc4NBXGV: not mounted
losetup: /dev/loop2: detach failed: No such device or address

sujitrp, 2018-07-27:
Am i doing some basic mistake ? guide me to correct the same

maywaiting, 2018-09-26:
@framps  Thanks for your fork. It works!!

framps, 2018-09-26:
Glad it works for you :smile: .

There is even a much more sophisticated version available. I frankly don't have neither enough corrupted images nor time to test all the different error and recovery scenarios.

That's why you're on your own if you use [this script](https://github.com/framps/PiShrink/blob/offer_filesystem_repair_option/pishrink.sh). But it works on a copy only and will not destroy your original image. Just use option -r to get similar recovery functionality or add option -p to enable paranoia recovery for pathological cases.

sujitrp, 2018-09-27:
@maywaiting pls guide us on how I made it  work for u

tonynieman, 2018-10-07:
Moin,
ich versuche schon länger mein Pi Image zu verkleinern um endlich ein Backup auf eine kleine SD-Karte zu schieben. Da ich mit dem original skript einen fehler ausgegeben bekommen habe, habe ich deines probiert. Dort bekomme ich leider ebenfalls einen Fehler ausgegeben und komme nicht weiter. 

pishrink_repair.sh: Gatherin data...
resize2fs 1.42.5 (29-Jul-2012)
Bitte zuerst âe2fsck -f /dev/loop0â laufen lassen.

pishrink_repair.sh: ERROR occured in line 285: resize2fs failed with rc 0

framps, 2018-10-07:
Great you managed to fix the issue and share how you fixed it :+1: . Unfortunately you didn't use option `-d` which creates some debug information and may help to debug the issue :-(. But nevertheless it's interesting you fixed the issue with
```
losetup -P /dev/loop0 raspbian.img
e2fsck -f /dev/loop0
```
Looks like this is something everybody facing issues with pishrink should try first.

stefanvictora, 2019-02-21:
Thanks for the suggested fix, I finally got it working for me as well after I specified the correct partition for `e2fsck` to use:

~~~
sudo losetup -P /dev/loop0 raspbian.img
sudo e2fsck -f /dev/loop0p2
-------------------------^ # note the added 'p2' to reference the second partition
~~~

See: https://raspberrypi.stackexchange.com/a/78361

And finally detaching the image again via `sudo losetup -d /dev/loop0` before running PiShrink.
This fixed the issue for me.

staehler, 2019-03-04:
@stefanvictora thanks for your hint.
It seems that e2fsck unmounts the /dev/loop0 device in case of filesystem repairs. I just needed to substitute 

```
#Make sure filesystem is ok
e2fsck -p -f "$loopback"
```
with
```
#Make sure filesystem is ok
e2fsck -y -f "$loopback"
if ! losetup $loopback &>/dev/null; then
  loopback=$(losetup -f --show -o $partstart "$img")
fi
```
to mount it again if it isn't. This did it my case.
Maybe someone finds this helpfull.

staehler, 2019-03-04:
> Great you managed to fix the issue and share how you fixed it 👍 . Unfortunately you didn't use option `-d` which creates some debug information and may help to debug the issue :-(. But nevertheless it's interesting you fixed the issue with
> 
> ```
> losetup -P /dev/loop0 raspbian.img
> e2fsck -f /dev/loop0
> ```
> Looks like this is something everybody facing an issue with pishrink should try first.

#56 might fix this issue in case you want to integrate this into your code. Thanks for this very usefull tool.

whatsupskip, 2019-05-08:
I think it would be really good if the repair option also included an option not create a copy. It could ask if you have made a backup first. I find I keep running out of storage space on the media I am trying to use to do the repair and then shrink.

easyyu, 2019-05-22:
> I've had a similar issue. I've created a image file from a working Raspbian SDCard using Win32DiskImager. When I ran pishrink, the tool gave me the "orphaned inode list" error. So I followed the [...]

Thx mate..
This was only solution what worked for me!

ejonesnospam, 2019-10-15:
> I've had a similar issue. I've created a image file from a working Raspbian SDCard using Win32DiskImager. When I ran pishrink, the tool gave me the "orphaned inode list" error. So I followed the [...]

Yes, helped me. Thanks. I vote to add this to pishrink

framps, 2019-10-15:
@gmenezesg 

> I've had a similar issue. I've created a image file from a working Raspbian SDCard using Win32DiskImager. When I ran pishrink, the tool gave me the "orphaned inode list" error. So I followed the [...]

Thank you very much for this fix :+1: . Why didn't you create a PR so that @Drewsif can easily integrate your fix into the mainstream :wink: ?

Kasperx, 2020-04-30:
@stefanvictora 
Thank you, that was my solution. Seemed like my 2nd partion had errors that had no effect on system, but pishrink could not handle it. After those 2 cmds the error got solved and the script could do its job.

Might be a great idea to include `e2fsck` to pishrinks script to solve future errors. All values neccessary for that are available at that point.

Drewsif, 2020-06-17:
Closing the old issue. `e2fsck` is now run by default!

closed 2020-06-17 by Drewsif