Drewsif/PiShrink #60: fails with 8gb image of Stretch lite "parted: invalid token: primary"

themzlab, 2018-04-09:
log:
mark@mark-Inspiron-1525 /temp $ sudo pishrink.sh stretch.img
Creating new /etc/rc.local
rootfs: 69461/465392 files (0.1% non-contiguous), 551275/1928192 blocks
resize2fs 1.42.13 (17-May-2015)
resize2fs 1.42.13 (17-May-2015)
Resizing the filesystem on /dev/loop4 to 770932 (4k) blocks.
Begin pass 2 (max = 27468)
Relocating blocks             XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
Begin pass 3 (max = 59)
Scanning inode table          XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
Begin pass 4 (max = 6792)
Updating inode references     XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
The filesystem on /dev/loop4 is now 770932 (4k) blocks long.

parted: invalid token: primary
Error: Expecting a file system type.
Shrunk stretch.img from 7.4G to 0

-------------------------------------------------------------------------------------------------------------
I created the image with the line:
sudo dd if=/dev/mmcblk0p2 bs=4M of=stretch_3.5.img

after unmounting the two partitions of the sd card using gparted.

closed 2018-04-09 by themzlab

John-Mc, 2018-04-09:
@themzlab - I've never used the `bs=4M` part when I create an image. What is the advantage over the default 512 bytes?

themzlab, 2018-04-09:
@John-Mc  the only reason I used it is that that parameter existed in the example I copied.  The advantage would be or could be increased speed in the reading or writing of the data.  If this isn't something one does everyday then I don't think speed would matter much and that part can be omitted.