Drewsif/PiShrink #61: Option to output to different path

heikoh81, 2018-05-01:
As far as I understand the script there is an option to create a copy of the original file and then pishrink works on this copy.
This, however, creates high simultaneous read/write operations on one HDD (slow, not good for the hardware).

What I would like to have is to specifiy an input file (that will not be changed) and a different output file (that will be created with shrinked size).
This way, I would have only read operations on HDD1, and write opeations on HDD2 (faster, and good for the hardware).

Is this possible?

Drewsif, 2018-05-02:
If the path is on another hard drive then it will read from the one the write it to the other. For example `pishrink.sh /home/me/saved.img /mnt/external/shrunk.img` will copy it and do the work on the other drive. It does copy the entire file then work on it, but there is not a good way to avoid that sadly.

Does that answer your question?

heikoh81, 2018-05-03:
Thanks for your answer, but my question remains open.

I understand the original syntax (which I tried, of course).
But as you describe, the file ist copied first, than worked on.
The copy-process is ok, but the working process not, as it needs a lot simulataneous read/write operations, thus putting the target drive under a lot of stress.
In regards to hard drive life, it would be much better if one drive only reads and the other only writes...
But if the used commands need to work on the file directly, than there probably is no way around this.

(Ramdisk is (not yet) an option, as I'm using 16GB SD-Cards in my Raspis and even on my main PC I only have 16 GB RAM...)

heikoh81, 2018-05-04:
Ok, I understand and it is ok then.

closed 2018-05-04 by heikoh81