Drewsif/PiShrink #68: PiShrink Fails running in WSL Ubuntu 1804

TheColonel2688, 2018-07-23:
I have installed PiShrink on WSL Ubuntu 18.04. I tried to run it with 'sudo pishrink.sh RPi-backup-Hipot-18-07-19-shrunk.img' . When I run it I receive the following error.

`Kernel not configured for semaphores (System V IPC). Not using udev synchronisation code.
losetup: cannot find an unused loop device
losetup: : failed to use device: No such device
tune2fs: No such file or directory while trying to open
Couldn't find valid filesystem superblock.
losetup: : failed to use device: No such device
mount: /tmp/tmp.vgF8KzVyQs: unknown filesystem type.
losetup: : failed to use device: No such device
md5sum: /tmp/tmp.vgF8KzVyQs/etc/rc.local: No such file or directory
/usr/local/bin/pishrink.sh: line 78: [: !=: unary operator expected
umount: /tmp/tmp.vgF8KzVyQs: not mounted.
losetup: : failed to use device: No such device
ext2fs_check_if_mount: Can't check if filesystem is mounted due to missing mtab file while determining whether  is mounted.
e2fsck: No such file or directory while trying to open
Possibly non-existent device?
losetup: : failed to use device: No such device
resize2fs 1.44.1 (24-Mar-2018)
ext2fs_check_mount_point: Can't check if filesystem is mounted due to missing mtab file while determining whether  is mounted.
ERROR: Image already shrunk to smallest size
losetup: : failed to use device: No such device
losetup: : failed to use device: No such device`

shatteredsword, 2018-07-25:
I don't think this script has ever worked under WSL unfortunately

timekills, 2020-06-21:
Debian or Ubuntu (and likely other images) fail in WSL2 with pishrink because pishrink looks for the /etc/mtab file to determine mounts.

Before I explain this, the quick solution (assuming you've already mounted the drive where your image is stored) is to **create a symlink between the /proc/self/mounts file and /etc/mtab**

`sudo ln -s /proc/self/mounts /etc/mtab`

Once this is done, you should not get the "ext2fs_check_if_mount: Can't check if filesystem is mounted due to missing mtab file while determining whether /dev/loop0 is mounted" error any more.

-----------------------------
Short background:
the mtab file (/etc/mtab) is a legacy from Unix to store mounted locations. It was included in many (most) Linux distros to ensure backward compatibility and Posix/Unix compatibility. More recently, most Linux distros create a symlink between the replacement file (/proc/self/mounts) and the legacy /etc/mtab.
Windows Subsystem for Linux (1 and 2) do not include that symlink, which is why you must create yourself.

kaszeta, 2020-10-28:
To my pleasant surprise, having just upgraded my WSL to WSL2 and installing a fresh Ubuntu 20.04 image... This worked out of the box.

crosland, 2021-01-30:
Hmmm, I have WSL2 with Ubuntu 20.04 but it doesn't "work out of the box" :(

```
pishrink.sh: Gathering data ...
Kernel not configured for semaphores (System V IPC). Not using udev synchronisation code.
Kernel not configured for semaphores (System V IPC). Not using udev synchronisation code.
losetup: cannot find an unused loop device
tune2fs: No such file or directory while trying to open
Couldn't find valid filesystem superblock.
tune2fs 1.45.5 (07-Jan-2020)
pishrink.sh: ERROR occurred in line 289: tune2fs failed. Unable to shrink this type of image

```

The sumlink mentioned in an earlier comment already exists
```
~/PiShrink$ ls -al /etc/mtab
lrwxrwxrwx 1 root root 19 Apr 23  2020 /etc/mtab -> ../proc/self/mounts

```

crosland, 2021-01-30:
It seems WSL2 does not support loopback

```
~/PiShrink$ modprobe loop
modprobe: FATAL: Module loop not found in directory /lib/modules/4.4.0-19041-Microsoft
```
@kaszeta did you do anything else to make it work?

timekills, 2021-02-01:
@crosland Did you try my work around above?

crosland, 2021-02-01:
Yes, my WSL already had a symlink but it pointed to **..**/proc/self/mounts. I changed it to /proc/self/mounts per your workaround and it still didn't work.

crosland, 2021-02-01:
No, I can't update :(
I've tried to find instructions on how to do it, but all I have found leads me to think it's not possible unless I join the Windows insider program.
I did download the .cab for 5.4.72 (that was an exercise in itself), extract the .msi and run it but nothing changed. I am still running the 4.4.0-19041-Microsoft  kernel
Some source refer to wsl --update but that is not a valid command.
My Windows build is 19041.746 It seems there is an update available to 20H2 so I will try downloading that.

crosland, 2021-02-01:
So Windows is updated to 20H219042.746 but I still don't know how to update the kernel.

kaszeta, 2021-02-01:
> So Windows is updated to 20H219042.746 but I still don't know how to update the kernel.

Okay, this we should be able to solve.

1. Go to the link above, and download and install either of the current WLS2 kernels there.
2. You need to convert your current WSL to use that new kernel.  I'm probably over-specifying the steps here, but 
- Open up a Powershell as root
- wsl --set-default-version 2
- wsl --set-version Ubuntu-20.04 2
- wsl --shutdown
- wsl
- Verify the new kernel with uname -a

I just completed these steps to update my 20.04 WSL installation to the 5.4.72 kernel and re-ran PiShrink to verify it still works.

crosland, 2021-02-01:
Thanks for sticking with me!

I still see the old kernel (transcript below)

To install the new kernel I simply double clicked the .msi from Windows Explorer, the same as I would for other software. Is that correct? Do I need to do it from a particular directory? Or some different way?

```
PS C:\WINDOWS\system32> wsl --set-default-version 2
For information on key differences with WSL 2 please visit https://aka.ms/wsl2
PS C:\WINDOWS\system32> wsl --set-version Ubuntu-20.04 2
Conversion in progress, this may take a few minutes...
For information on key differences with WSL 2 please visit https://aka.ms/wsl2
Please enable the Virtual Machine Platform Windows feature and ensure virtualization is enabled in the BIOS.
For information please visit https://aka.ms/wsl2-install
PS C:\WINDOWS\system32> wsl --shutdown
PS C:\WINDOWS\system32> wsl
andrew@SPROG2-PC:/mnt/c/WINDOWS/system32$ uname -r
4.4.0-19041-Microsoft
andrew@SPROG2-PC:/mnt/c/WINDOWS/system32$
```

kaszeta, 2021-02-01:
Hmmm.  I'll have to look into that one.

The line "Please enable the Virtual Machine Platform Windows feature" makes me wonder if everything isn't fully installed for you.

crosland, 2021-02-02:
I googled that message but I'm not sure I fully understand the implications. I'm mostly looking at https://github.com/microsoft/WSL/issues/5363

Powershell still has my command history (up-arrow) and I can see that I did (some time ago) run the command
`dism.exe /online /enable-feature /featurename:VirtualMachinePlatform /all /norestart
`
But in the performance tab of Task Manager I see
```
Virtualization: Disabled
Hyper-V support: Yes

```
and in Windows Features I see Hyper-V unticked and Virtual Machine Platform ticked.

I see from 
`systeminfo`
that virtualization is not enabled in firmware.

I am unsure exactly what combination I need. Some talk about "turning it off and on again" for the BIOS setting.

At this point I'll add the comment and reboot into the BIOS.

crosland, 2021-02-02:
It works! Thank you for all the help.

After enabling virtualization in the BIOS and going through the above steps again I now have the recent kernel and the script works.

The only mention of enabling the BIOS setting in the Microsoft instructions https://docs.microsoft.com/en-us/windows/wsl/install-win10 is if you get an error message. I had no errors and everything seemed to work.

Two downsides:
The VM kills the machine performance by hogging all the memory, e.g. when moving the image from Windows filesystem to WSL file system.

The script is not as effective as Andrew Oakley's https://github.com/aoakley/cotswoldjam/tree/master/raspbian-shrink
pishrink reduced a 15G image to 6G, raspian-shrink to 4.3G With raspian-shrink I have to manually setup a crontab entry in the image, before shrinking, to do the filesystem resize on first boot, but it's worth it to get the smaller image.

Hopefully, raspian-shrink will now also work on WSL :)

polystyreneMan, 2021-05-26:
Thanks, guys. The WSL 2/update process worked great for me as well.

Drewsif, 2021-06-11:
> The script is not as effective as Andrew Oakley's https://github.com/aoakley/cotswoldjam/tree/master/raspbian-shrink
> pishrink reduced a 15G image to 6G, raspian-shrink to 4.3G With raspian-shrink I [...]

The good news for you is that once I implement #59 (will be the same as in raspbian-shrink) both scripts should reduce to the same size. Pishrink does add a bit of extra fluff (a couple MB worth) since it made some images unbootable with that little of disk space remaining.

hayden-t, 2022-06-13:
If running virtual machine with virtualbox guest additions, just copy the image to local internal system first.

KimJuyeop, 2023-01-18:
jyk@LAPTOP-UPVHNNDL:/mnt/c/rasp_BackUp$ sudo ./pishrink.sh raspBackup.img
pishrink.sh v0.1.2
pishrink.sh: Gathering data ...
tune2fs: Bad magic number in super-block while trying to open /dev/loop0
tune2fs 1.44.1 (24-Mar-2018)
/dev/loop0 contains a vfat file system
pishrink.sh: ERROR occurred in line 297: tune2fs failed. Unable to shrink this type of image
------------------------------------------------------------------------------------------------------
help Me

KimJuyeop, 2023-01-18:
jyk@LAPTOP-UPVHNNDL:/mnt/c/rasp_BackUp$ sudo ./pishrink.sh raspBackup.img

--------------------------------
pishrink.sh v0.1.2
pishrink.sh: Gathering data ...
tune2fs: Bad magic number in super-block while trying to open /dev/loop0
tune2fs 1.44.1 (24-Mar-2018)
/dev/loop0 contains a vfat file system
pishrink.sh: ERROR occurred in line 297: tune2fs failed. Unable to shrink this type of image
-------------------------------------------------------------------------------------------------------------help plz

closed 2023-02-07 by Drewsif