Drewsif/PiShrink #69: New feature to control size of new image

22nra, 2018-07-23:
Many thanks for this excellent utility, is it possible to add a feature to control the size of the new image.
Maybe add a -g option as below
pishrink -g3 imagetobesized.img
This would resize the image to 3gb, the -g would of course stop any expansion as used with the -s option
Another idea could be -b where you could control the size of image in 4k blocks.
thanks and regards :)

Drewsif, 2018-07-26:
Why would you want to do this over just shrinking to the smallest size? For when the auto resize is not used?

22nra, 2018-07-28:
Hi and thanks for responding to this feature request, sorry but my reply appears to be heading off topic. 
I've been working on a Pi project for some time and initially my image was about 1.7gb. I would then shrink and run my project off 2gb cards which gave me slight overhead and still working with a small images. My image has now increased over 2gb so I now use 4gb cards which are becoming gradually less available and more costly pushing me sometimes to use cheaper 8gb cards for a 2gb image. My goal is to continue to work with a small image which keeps any backups small in size. I stupidly thought that if I burned my image without resize the image would stay at the original size. But if I take a backup using a tool like Win32DiskImager the image resizes to the size of the card.
IF, I could find clone/backup tool that would copy just the size of the image my request would have some sense. 
Another card issue if of course the dreaded SD corruption. Most (some?) cards cater for any errors automatically and reallocate bad blocks allowing the card to work fine with block errors. If an 8gb SD card had a 3gb image would the card cope better with corruption (way off topic). 
Hope this ramble has answered your reply but until a get a better clone tool the feaure has no value.

slappymcphee, 2018-08-13:
I personally would like the idea of the feature enhancement. When I shrink it would be nice if I could allocate say an additional 500MB in case my expand scripts do not work to make minor edits or give the ability to add additional small files post-shrink via Paragon extFS partition mounting.

Drewsif, 2020-06-17:
The script already adds a bit of space to the end for this purpose. Unless you startup procedure is creating a large amount of files you should be able to boot and expand manually if you need.

I believe you're correct @22nra that without a backup tool that smartly looks at the partition table the feature wouldn't really help your use case.

closed 2020-06-17 by Drewsif