Drewsif/PiShrink #73: Raspbian not completely booting after pishrink

marcert, 2018-08-24:
Hi,
i created a new imagefile from raspbian (all updates installed) with rpi-clone and tested it. Works.
Then I tried to shrink it with PiShrink (file was then around 5G) wrote this to the SD.
If I now try to boot with this SD-Card I see the 4 Raspberries, then the splash screen and then no graphical UI.
I can also ssh to the Raspberry and see that e.g. the lightdm process is running but I do not get the normal UI via HDMI and also not via xrdp.
Has somebody an idea what happens there? What can I do to change that behavior?

Thanks and Kind Regards,
Marco

sealyau, 2019-01-09:
what services as listed as failed when you run "systemctl --state=failed"

Drewsif, 2020-06-17:
Closing stale issue, feel free to re-open with more details

closed 2020-06-17 by Drewsif

davidzfr, 2023-11-23:
Hi, I've the same problem.
Raspbian 12 desktop 32 bits.
No GUI, no access to desktop on the screen.
Only an underscore blinking at the top left corner of the screen.
I can access to the Rpi through ssh.
The config.txt is exactly the same.
Any Idea ?