Drewsif/PiShrink #74: Raspbian stretch lite - issue with shrinking

rafek00, 2018-08-28:
After making copy od raspbian with command:
dd if=/dev/sda of=/mnt/share/rpi.img

i have issue with shrinking:
root@raspberrypi:~# ./pishrink.sh /mnt/share/rpi.img
rootfs: 54900/237120 files (0.2% non-contiguous), 782366/955776 blocks
resize2fs 1.43.4 (31-Jan-2017)
ERROR: Image already shrunk to smallest size

What i made wrong ?

sickboy2711, 2019-02-03:
Hello, I have the same problem
When I run, I get this message:
`root @ rpi-Pi-Ager: / home / pi # sudo /usr/local/bin/pishrink.sh -s / home / pi / backup / be tarelease_Denni.img
Skipping autoexpanding process ...
rootfs: 47483/118560 files (0.2% discontinuous), 371045/471040 blocks
resize2fs 1.43.4 (31-Jan-2017)
ERROR: Image already shrunk to smallest size
`
I use: Rasbian Jessy Lite and have mounted my NAS

Drewsif, 2020-06-17:
Closing this as a duplicate. I will be looking into this issue

closed 2020-06-17 by Drewsif