Drewsif/PiShrink #76: Using PiShrink with resized partition

bizzar721, 2018-09-15:


Drewsif, 2018-10-30:
The only reason I can think it would break is if it shrinks too much there wont be free space to boot? But I think it should work, what was the error you were getting?

bizzar721, 2018-10-31:
I created another backup of my expanded build (Retropie 4.4 Stretch) so I can test.  When run with command `sudo ./pishrink.sh expandedbackup.img`  I get 
`Creating new /etc/rc.local
retropie: 536767/7488992 files (0.3% non-contiguous), 30344081/30704384 blocks
resize2fs 1.42.13 (17-May-2015)
ERROR: Image already shrunk to smallest size
`

Drewsif, 2020-06-17:
You're not the first one to report this issue, it's on my todo list so I'm going to close this issue as a duplicate of #80

closed 2020-06-17 by Drewsif