Drewsif/PiShrink #78: PIShrink on WSL (Debian GNU/Linux)

KraftMK, 2018-10-14:
Hello Drewsif,

i can't run PIShrink on the Windows Subsystem for Linux.
I guess it is a problem with the WSL Kernel.
Here the Command Line Text:

root@DESKTOP-FPR4TUC:/mnt/d# sudo pishrink.sh PI_IMG_VDS.img pi-shrink.img
Copying PI_IMG_VDS.img to pi-shrink.img...
Kernel not configured for semaphores (System V IPC). Not using udev synchronisation code.
losetup: cannot find an unused loop device
losetup: : failed to use device: No such device
tune2fs: No such file or directory while trying to open
Couldn't find valid filesystem superblock.
losetup: : failed to use device: No such device
mount: unknown filesystem type
losetup: : failed to use device: No such device
md5sum: /tmp/tmp.hIuCWGRTrr/etc/rc.local: No such file or directory
/usr/local/bin/pishrink.sh: line 78: [: !=: unary operator expected
umount: /tmp/tmp.hIuCWGRTrr: not mounted
losetup: : failed to use device: No such device
ext2fs_check_if_mount: Can't check if filesystem is mounted due to missing mtab file while determining whether  is mounted.
e2fsck: No such file or directory while trying to open
Possibly non-existent device?
losetup: : failed to use device: No such device
resize2fs 1.43.4 (31-Jan-2017)
ext2fs_check_mount_point: Can't check if filesystem is mounted due to missing mtab file while determining whether  is mounted.
ERROR: Image already shrunk to smallest size

Thank you for your help.

best regards
MKraft

xiegeo, 2018-10-16:
As far as I know, WSL is not supported. The workaround I use is put the image on network share and run PiShrink from a rpi or VM.

Drewsif, 2018-10-30:
I can definitely say that WSL is not currently supported. I would be surprised if I could get it working but at some point I will try

closed 2018-10-30 by Drewsif

robertf26, 2020-06-10:
I successfully ran PiShrink on WSL 2 with the Debian distro. All I had to do was run `ln -s /proc/mounts /etc/mtab`
before running the script.

Gh0stWalk3r, 2021-02-02:
Big thanks to @robertf26 for the hint!
I can confirm that it works with Debian on WSL 2.
Full procedure is:

1. [Install WSL 2](https://docs.microsoft.com/en-us/windows/wsl/install-win10)
2. [Install and start Debian](https://www.microsoft.com/store/productId/9MSVKQC78PK6). You will need to define a user and password
3. Run `sudo apt update && sudo apt -y dist-upgrade`
4. Run `sudo apt install -y git parted`
5. Run `git clone https://github.com/Drewsif/PiShrink.git`
6. Run `sudo chmod +x PiShrink/pishrink.sh`
7. Run `sudo cp PiShrink/pishrink.sh /usr/local/bin`
8. Run `sudo ln -s /proc/mounts /etc/mtab`
9. Now you can shrink your image, e.g. `sudo pishrink.sh bigFile.img smallFile.img`

Ic33s, 2021-02-24:
Unfortunatly I've got no luck with it, maybe anyone has an Idea what's going wrong. Thank you in advance:

1. Created my input img by using Win32DiskImager.
2. Have done the steps from @Gh0stWalk3r 
3. Got an error:
`user@Desktop:/mnt/f/_BackupSDs$ sudo pishrink.sh OctoPrint_Ender3pro.img OctoPrint_Ender3pro_shrinked.img
pishrink.sh v0.1.2
pishrink.sh: Copying OctoPrint_Ender3pro.img to OctoPrint_Ender3pro_shrinked.img... ...
pishrink.sh: Gathering data ...
/dev/mem: No such file or directory
Kernel not configured for semaphores (System V IPC). Not using udev synchronisation code.
/dev/mem: No such file or directory
Kernel not configured for semaphores (System V IPC). Not using udev synchronisation code.
losetup: cannot find an unused loop device
tune2fs: No such file or directory while trying to open
Couldn't find valid filesystem superblock.
tune2fs 1.44.5 (15-Dec-2018)
pishrink.sh: ERROR occurred in line 289: tune2fs failed. Unable to shrink this type of image`

Ic33s, 2021-02-24:
Unfortunatly I've got no luck with it. Maybe anyone has an idea what's going wrong. 

Thank you in advance.

Created my input img by using Win32DiskImager.
Have done the steps from @Gh0stWalk3r
Got an error:
user@Desktop:/mnt/f/_BackupSDs$ sudo pishrink.sh OctoPrint_Ender3pro.img OctoPrint_Ender3pro_shrinked.img
pishrink.sh v0.1.2
pishrink.sh: Copying OctoPrint_Ender3pro.img to OctoPrint_Ender3pro_shrinked.img... ...
pishrink.sh: Gathering data ...
/dev/mem: No such file or directory
Kernel not configured for semaphores (System V IPC). Not using udev synchronisation code.
/dev/mem: No such file or directory
Kernel not configured for semaphores (System V IPC). Not using udev synchronisation code.
losetup: cannot find an unused loop device
tune2fs: No such file or directory while trying to open
Couldn't find valid filesystem superblock.
tune2fs 1.44.5 (15-Dec-2018)
pishrink.sh: ERROR occurred in line 289: tune2fs failed. Unable to shrink this type of image

gt360, 2021-06-18:
>        pishrink.sh: ERROR occurred in line 289: tune2fs failed. Unable to shrink this type of image

Same here.
Solution: install wsl2 and upgrade your linux distribution.
Using latest (20H2) W10 is not enough, wsl is not upgrading itself with windows upgrades: https://docs.microsoft.com/en-us/windows/wsl/install-win10

kaltinril, 2021-09-08:
@Gh0stWalk3r Thanks for the information, followed your steps exactly and they work.  @Drewsif  should put this on the main README for how to install/use it for WSL2

PietroPizzi69, 2022-03-24:
Has something changed since last year? It doesn't work for me. I think I have done it like it should be!?

I have "Windows 10 Pro 21H2 19044.1586" (Up-to-date).

>wsl --status
Standardversion: 2
(I don't have to upgrade anything because I installed WSL for the first time and got v2 as standard.)

I tested 2 Dists, Debian and Ubuntu. In both I have done the usual 'sudo apt update && sudo apt -y upgrade'.
In both I installed the pigz package (for -z).
In Debian I installed git and parted as described and set the /etc/mtab link (In Ubuntu the mtab link exists).
I copied the pishrink.sh in the PATH and set it as executable.

I tried (in both Dists) to shrink an image (.img not compressed) in the Win. Desktop folder (with -zap and without any switches), with an cope ('pishrink.sh in.img out.img') and diretk ('pishrink.sh in.img'). I also tried to copy the image inside the WSL path of both Dists ("\\wsl$\<dist>\home\<user>\Images\").

I always get the same result:
pishrink.sh v0.1.2
pishrink.sh: Gathering data ...
Kernel not configured for semaphores (System V IPC). Not using udev synchronisation code.
Kernel not configured for semaphores (System V IPC). Not using udev synchronisation code.
losetup: cannot find an unused loop device
tune2fs: No such file or directory while trying to open
Couldn't find valid filesystem superblock.
tune2fs 1.43.4 (31-Jan-2017)
pishrink.sh: ERROR occurred in line 297: tune2fs failed. Unable to shrink this type of image

Does anybody know what is wrong?

wauwau0977, 2022-05-24:
Hi Guys,
I did quite the same as @PietroPizzi69 . Also for me, it's not working on WSL2, neither on debian nor ubuntu LTS. 
Cheers,

Shiny380, 2023-01-25:
@wauwau0977 @PietroPizzi69 Had the same issue
Check your specific distro WSL version with 
`wsl --list --verbos` in powershell. Mine were version 1 despite having WSL 2 installed
Solution for me:
- `wsl --set-version <Distro> 2`
OR
- uninstall your distro
- `wsl --set-default-version 2`
- reinstall distro

Cred for this: https://www.youtube.com/watch?v=77vOeofG0Kg

sebther, 2023-03-12:
Hi all!

Where do I have to store my .img for running according to step 9?
#_"9. Now you can shrink your image, e.g. sudo pishrink.sh bigFile.img smallFile.img"_

Thanks - Helmut

stevennoyce, 2024-08-15:
I wanted to note that PiShrink has been working great for me under the Ubuntu WSL distro, as long as it is WSL 2.  Thanks so much for a great package and for the tips on this thread of making sure to use WSL 2.  I know others have had success with the Debian distro under WSL, but I wanted to add my experience to the conversation that it works great for me under Ubuntu WSL 2 natively (no special setup, I just needed to run `wsl --set-version Ubuntu 2` in powershell).