Drewsif/PiShrink #79: Using pshrink on os x

phototobi, 2018-10-22:
Is it possible to use phrink on os x. Maybe by exchanging parted with gparted as it is not available for os x?

Thanks in advance

i-ghost, 2019-05-03:
I don't believe OSX/macOS can mount ext2/3/4 natively, so you'd need something like osxfuse.

Instead of messing about with FUSE, I've had success using Docker for Mac.

You'll first need to allocate sufficient storage to the VM using the Docker for Mac settings panel.

Assuming `source.img` is in the current working directory:

```shell
docker run --privileged --rm -it -v ${PWD}:/shrink ubuntu:18.04
apt-get install git parted
git clone https://github.com/Drewsif/PiShrink.git
export PATH=/PiShrink/:${PATH}
cd /shrink
pishrink.sh source.img out.img
```

So if anything, this repo could benefit from a Dockerfile or docker instructions.

Drewsif, 2020-06-17:
There is no native way but I do want to create docker directions for both mac and windows soon(tm)

closed 2020-06-17 by Drewsif