Drewsif/PiShrink #81: /dev/root does not exist or is not a symlink

AlfaBravoX, 2018-11-05:
When using raspi-config to again resize filesystem after using this script, it cannot be done: /dev/root does not exist or is not a symlink. Don't know how to expand . Creating link manually does not help either. any idea?

Drewsif, 2018-11-05:
What distro are you using?

AlfaBravoX, 2018-11-05:
Raspbian 
cat /etc/debian_version
9.4
uname -a 4.9.64-v7+ #1053 SMP Tue Nov 21 14:56:27 GMT 2017 armv7l GNU/Linux

AlfaBravoX, 2018-11-05:
OK i resized partition by fdisk manually, then resize2fs /dev/mmcblk0p2 and it looks good:
 df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root        15G  6.3G  7.3G  47% /

not sure if this relates to the script, the error message of raspi-config still persists

Drewsif, 2019-02-15:
I will look into it

Drewsif, 2020-06-17:
Closing for now as I've not been able to reproduce and have made a ton of changes since. If the newest version of PiShrink is giving you an issue please re-open this

closed 2020-06-17 by Drewsif