Drewsif/PiShrink #82: issues with copy option

montaqa, 2018-11-15:
Hi,

if i specify the newimagefile.img parameter, the following errors is thrown:

```
oracle@oracle-Veriton-L670G:~$ sudo pishrink.sh -s pi.img pi_small.img
[sudo] password for oracle:
Copying pi.img to pi_small.img...
Skipping autoexpanding process...
rootfs: recovering journal
rootfs: 39688/942480 files (0.1% non-contiguous), 330827/3877248 blocks
resize2fs 1.44.1 (24-Mar-2018)
resize2fs: Invalid argument while trying to open /dev/loop21
/usr/local/bin/pishrink.sh: line 151: [[: Couldn'tfindvalidfilesystemsuperblock.                                                                             : syntax error: invalid arithmetic operator (error token is "'tfindvalidfilesyst                                                                             emsuperblock.")
/usr/local/bin/pishrink.sh: line 157: 3877248 - Couldn'tfindvalidfilesystemsuper                                                                             block.: syntax error: invalid arithmetic operator (error token is "'tfindvalidfi                                                                             lesystemsuperblock.")
resize2fs 1.44.1 (24-Mar-2018)
resize2fs: Invalid argument while trying to open /dev/loop21
Couldn't find valid filesystem superblock.
ERROR: resize2fs failed...
mount: : mount point does not exist.
mv: cannot stat '/etc/rc.local.bak': No such file or directory
umount: : no mount point specified.
losetup: /dev/loop21: detach failed: No such device or address

```
without this option, process is running without errors. image is created by Win32DiskImager

Drewsif, 2019-02-15:
That odd... it must have something weird to do with your system. The copy just uses `cp` nothing that should error it out. If you're using virtualbox shared folders there are issues with it. Try coping it into the VM if thats how you're running it

closed 2019-02-15 by Drewsif

fofles, 2019-02-15:
I had the same issue .
 VM Linux  with shared folders from the host windows.  
My workaround  was to use as the name of the new image an existing file previously made with Windows.  

If I did a new file in VM Linux,  touch  test.img ; chown 777  ,  and I used it to the new image name,  the same issue. 
 
I hope this help.