Drewsif/PiShrink #84: Script runs but does not shrink

svenyonson, 2018-12-13:
I have a 32GB image made with Apple Pi Baker. I ran the script on Ubuntu 18.04 and got the following: 
```
sudo pishrink.sh 32GB.img
Creating new /etc/rc.local
root: 153383/1839600 files (0.2% non-contiguous), 1335211/7352448 blocks
resize2fs 1.44.1 (24-Mar-2018)
resize2fs 1.44.1 (24-Mar-2018)
Resizing the filesystem on /dev/loop7 to 1674987 (4k) blocks.
Begin pass 2 (max = 535573)
Relocating blocks             XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
Begin pass 3 (max = 225)
Scanning inode table          XXXXXXXXXXXXXXcalXXXcXXXXXXXXXXXXXXXXXXXXXXX
Begin pass 4 (max = 14289)
Updating inode references     XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
The filesystem on /dev/loop7 is now 1674987 (4k) blocks long.

Error: You requested a partition from 1799356416B to 8660103168B (sectors 3514368..16914264).
The closest location we can manage is 31914983424B to 31914983424B (sectors 62333952..62333952).
Shrunk 32GB.img from 30G to 30G
```

Drewsif, 2019-02-15:
I've not looked into Apple Pi Baker to see if its doing anything weird but I'll check it out

csik, 2019-05-24:
Same running an Ubunto 18.04 vagrant on os x. Exact same size.

maHecht, 2019-08-20:
Same on Lubuntu 16.04 in a VM. Are there any news in this Topic?

audunhy, 2019-08-23:
I had a similar issue on CentOS 7. I fixed it by changing line  264 adding " around $endresult.
```
264c264
< endresult=$(tail -1 <<< $endresult | cut -d ':' -f 2 | tr -d 'B')
---
> endresult=$(tail -1 <<< "$endresult" | cut -d ':' -f 2 | tr -d 'B')

```

closed 2020-06-17 by Drewsif