Drewsif/PiShrink #85: Not working with Chinese computer

MistEO, 2018-12-29:
$ sudo ./pishrink.sh debug2.img 
[sudo] mreo 的密码： 
/dev/loop34：124661/214144 文件（0.7% 为非连续的）， 729259/895714 块
resize2fs 1.44.5 (15-Dec-2018)
./pishrink.sh: 行 150: [[: 预计文件系统的最小尺寸：890709: 语法错误: 需要操作数 (错误符号是 "预计文件系统的最小尺寸：890709")
./pishrink.sh: 行 156: 895714 - 预计文件系统的最小尺寸：890709: 语法错误: 需要操作数 (错误符号是 "预计文件系统的最小尺寸：890709")
resize2fs 1.44.5 (15-Dec-2018)
resize2fs: 无效的新大小： 预计文件系统的最小尺寸：890709

ERROR: resize2fs failed...
umount: /tmp/tmp.caildzARsd: target is busy.

MistEO, 2018-12-29:
In line 150
`minsize=$(resize2fs -P "$loopback" | cut -d ':' -f 2 | tr -d '     ')`

it split by semicolon ":", but my computer output is a full-width semicolon "：", which causes this error.

The solution is change line 150 to
`minsize=$(LANG='en_us' resize2fs -P "$loopback" | cut -d ':' -f 2 | tr -d '     ')`

then it will working

framps, 2018-12-29:
What about running pishrink in English?
```
sudo LANG=en_US ./pishrink.sh debug2.img
```

Drewsif, 2019-07-24:
Seems like setting the language in the script to english always would be helpful in debugging. Since most issues are posting in german and my german is very very bad at best

framps, 2019-07-24:
@Drewsif Totally agree with you. But don't hesitate to ping me with @framps if you need help with German bug reports :smile:

Drewsif, 2020-06-17:
Closing as a duplicate of #120

closed 2020-06-17 by Drewsif