Drewsif/PiShrink #86: PiShrink the SD from which I booted the Pi?

John-Mc, 2019-01-06:
I've been installing and running PiShrink from my Pi. My process is to first create an image of the current boot SD card using the command: 
`sudo dd if=/dev/mmcblk0 of=pathname/mynewimage.img`

Where "pathname" is the path to a USB drive attached to the pi.

I then run navigate to that directory and run PiShrink against the mynewimage.img file to create the shrunken version with the following command:
`sudo pishrink.sh mynewimage.img`

This works, but since the "sudo dd ..." command can take some time to run, I'm wondering if it's possible to run PiShrink directly against the boot SD card (and routing the output to the attached USB drive) without first copying it to an .img file. If this is possible, what would that command look like? I'm not too knowledgeable in this area, but I'm thinking of trying something like:
`sudo pishrink.sh /dev/mmcblk0 pathname/mynewimage.img`

Woud something like that work, or does PiShrink have to start with an .img file, rather than the card itself?

sealyau, 2019-01-10:
Hi buddy,
I wouldn't recommend that. Also are you mounting the root filesystem read only prior to creating the image file?
As for speeding up the process of you don't specify a new image file it won't first make a copy it will just work on the original image file.
Hope this helps

John-Mc, 2019-01-10:
If I'm running to create a shrunken backup of the Pi's current boot SD, I need to create the image at some point so I have something for the input to PiShrink. That was the whole point of my inquiry: can PiShrink make a shrunken image directly from the Boot SD, rather than first making an image of that SD. If you look at the commands I listed for my current workflow, I'm not creating a new image when running PiShrink, I'm running it on the image I have already created.

Also, I'm not following why I would need to mount the root file system as read only. The command I start with `sudo dd if=/dev/mmcblk0 of=pathname/mynewimage.img` is just a command that makes a backup image of the boot SD card (which includes the root file system). I've done it dozens of times without issue. I've never heard of write-protecting a drive before backing it up on any operating system. Can you explain to me what the risk is here?

sealyau, 2019-01-10:
Reason for recounting the file system read only prior to creating the image is that dd doesn't lock a file when reading that part of the disk and if another service updates the file at that point you may run into an issue where your back up image has a curupted file and won't boot or be buggy. Usually this isn't an issue as when backing up a drive it's not a system disk and it's recommended to unmount before backup.

This part of code would indicate your process would fail

#Args
img="$1"

#Usage checks
if [[ -z "$img" ]]; then
  usage
fi
if [[ ! -f "$img" ]]; then
  echo "ERROR: $img is not a file..."
  exit -2

You may be better off useing resize2fs to shrink your root partition then use dd to make an image which will be already shrunk.

Drewsif, 2019-01-10:
PiShrink could run the dd command for you, but since it uses resize2fs and other utilities that are destructive I don't want to run that on a live system. It would also require added code to undo the changes made to the live system which I think could get messy. I can look into having dd be run automatically though. I used cp over dd since you can save some time/space using sparse files
**TLDR;** It would have to run dd for you and you would save no time

To the read only point, the image has less of a chance to be corrupted if you take the image while the pi is offline, since blocks could be missing or not as intended if they are being written to while the backup is going

closed 2019-01-10 by Drewsif

John-Mc, 2019-01-10:
Thanks, folks!