Drewsif/PiShrink #88: shrink gives errors

bennybubble, 2019-01-11:
Hi,
After a problem with my rpi (bad sd card) I got a new SD card on which I wanted to restore a previously made backup (using raspibackup.sh).
Unfortunately the new sd-card (16 Gb) has slighty smaller space left and I needed to shrink the backup img.
However the shrik script gives me problems and don't know where it came from:

`bob@bob-VirtualBox:~/tmp$ sudo pishrink.sh -s oh2.img
[sudo] password for bob:            
Skipping autoexpanding process...
/dev/loop0: recovering journal
/dev/loop0: Clearing orphaned inode 2123 (uid=0, gid=0, mode=0100600, size=0)
/dev/loop0: Clearing orphaned inode 1120 (uid=0, gid=0, mode=0100600, size=0)
/dev/loop0: Clearing orphaned inode 612302 (uid=110, gid=115, mode=0100664, size=107324)
/dev/loop0: Clearing orphaned inode 8133 (uid=113, gid=119, mode=0100600, size=0)
/dev/loop0: Clearing orphaned inode 8121 (uid=113, gid=119, mode=0100600, size=0)
/dev/loop0: Clearing orphaned inode 7793 (uid=113, gid=119, mode=0100600, size=0)
/dev/loop0: Clearing orphaned inode 7791 (uid=113, gid=119, mode=0100600, size=0)
/dev/loop0: Clearing orphaned inode 7708 (uid=113, gid=119, mode=0100600, size=0)
/dev/loop0: Clearing orphaned inode 18704 (uid=0, gid=0, mode=0100755, size=24422673)
/dev/loop0: Clearing orphaned inode 10629 (uid=0, gid=0, mode=0100644, size=426976)
/dev/loop0: Clearing orphaned inode 10769 (uid=0, gid=0, mode=0100644, size=3132836)
/dev/loop0: Clearing orphaned inode 2413 (uid=0, gid=0, mode=0100644, size=300868)
/dev/loop0: Clearing orphaned inode 2412 (uid=0, gid=0, mode=0100644, size=1418540)
/dev/loop0: Clearing orphaned inode 27981 (uid=0, gid=0, mode=0100755, size=226940)
/dev/loop0: Clearing orphaned inode 15688 (uid=0, gid=0, mode=0100644, size=132620)
/dev/loop0: Clearing orphaned inode 1205 (uid=0, gid=0, mode=0100644, size=50652)
/dev/loop0: Clearing orphaned inode 1232 (uid=33, gid=33, mode=0100666, size=0)
/dev/loop0: Inodes that were part of a corrupted orphan linked list found.  

/dev/loop0: UNEXPECTED INCONSISTENCY; RUN fsck MANUALLY.
	(i.e., without -a or -p options)
resize2fs 1.44.1 (24-Mar-2018)
resize2fs: Invalid argument while trying to open /dev/loop0
/usr/local/bin/pishrink.sh: line 151: [[: Couldn'tfindvalidfilesystemsuperblock.: syntax error: invalid arithmetic operator (error token is "'tfindvalidfilesystemsuperblock.")
/usr/local/bin/pishrink.sh: line 157: 3916288 - Couldn'tfindvalidfilesystemsuperblock.: syntax error: invalid arithmetic operator (error token is "'tfindvalidfilesystemsuperblock.")
resize2fs 1.44.1 (24-Mar-2018)
resize2fs: Invalid argument while trying to open /dev/loop0
Couldn't find valid filesystem superblock.
ERROR: resize2fs failed...
mount: : mount point does not exist.
mv: cannot stat '/etc/rc.local.bak': No such file or directory
umount: : no mount point specified.
losetup: /dev/loop0: detach failed: No such device or address
bob@bob-VirtualBox:~/tmp$ 
`
I'm running Mint-Linux in VBox on win10  and copied the img into the guest-os.
Also tried other backups but same result.

Any hints greatly  appreciated!

-ben

awunnenb, 2019-01-13:
same issue with ubuntu and arch-linux

Drewsif, 2019-02-15:
So the problem is your filesystem had errors on it. Currently PiShrink does not automatically fix them for you, but I plan on adding that in

stefanvictora, 2019-02-21:
I had a similar problem and described my fix as a comment in this issue: https://github.com/Drewsif/PiShrink/issues/56#issuecomment-466088725 Maybe this helps. Regards!

bennybubble, 2019-02-27:
Thanks, it did !

closed 2019-02-27 by bennybubble