Drewsif/PiShrink #92: shrink gines errors #2

AlexanderShniperson, 2019-01-27:
After run at Ubuntu 18.04 i see this
```
user@user-pc:~$ sudo ./pishrink.sh /media/sf_raspy/relay_proj_200119.img
tune2fs: Input/output error while trying to open /dev/loop7
Couldn't find valid filesystem superblock.
mount: /tmp/tmp.pDOb40n21z: can't read superblock on /dev/loop7.
md5sum: /tmp/tmp.pDOb40n21z/etc/rc.local: No such file or directory
./pishrink.sh: line 78: [: !=: unary operator expected
umount: /tmp/tmp.pDOb40n21z: not mounted.
e2fsck: Operation not permitted while trying to open /dev/loop7
You must have r/w access to the filesystem or be root
resize2fs 1.44.1 (24-Mar-2018)
resize2fs: Operation not permitted while trying to open /dev/loop7
./pishrink.sh: line 151: [[: Couldn'tfindvalidfilesystemsuperblock.: syntax error: invalid arithmetic operator (error token is "'tfindvalidfilesystemsuperblock.")
./pishrink.sh: line 157: - Couldn'tfindvalidfilesystemsuperblock.: syntax error: invalid arithmetic operator (error token is "'tfindvalidfilesystemsuperblock.")
resize2fs 1.44.1 (24-Mar-2018)
resize2fs: Operation not permitted while trying to open /dev/loop7
Couldn't find valid filesystem superblock.
ERROR: resize2fs failed...
mount: /tmp/tmp.pDOb40n21z: can't read superblock on /dev/loop7.
mv: cannot stat '/tmp/tmp.pDOb40n21z/etc/rc.local.bak': No such file or directory
umount: /tmp/tmp.pDOb40n21z: not mounted.
losetup: /dev/loop7: detach failed: No such device or address
```
the given image for program is taken by `dd` command, the source of image is most fresh Raspbian lite

unclehowell, 2019-03-14:
Did you ever figure this?

AlexanderShniperson, 2019-03-15:
@unclehowell 
as i remember right this error happens when we try to shrink image from being virtual machine and joined hdd volume.

solution is: 
copy image into virtual machine then shrink.

penrods, 2019-06-05:
Can confirm, was able to do basically this in my VM of Ubuntu 18.04:
```bash
cp /media/sf_Shared/some.img /tmp/some.img
sudo ./pishrink.sh /tmp/some.img
mv /tmp/some.img /media/sf_Shared/some.img
```

The only problem is the additional time is making sure you have a big enough VM drive and then waiting for the copy and move on multi-gigabyte files.

Drewsif, 2020-06-17:
Unfortunately VirtualBox does not work as well as it should. I added a warning about this to the README

closed 2020-06-17 by Drewsif