Drewsif/PiShrink #95: "is not a file..."

lar3ry, 2019-02-18:
I'm probably missing something, but following the directions in the readme, when I tried to shrink, I got this:

pi@octopi:~ $ sudo pishrink.sh pi.img
[sudo] password for pi:
ERROR: pi.img is not a file...

John-Mc, 2019-02-18:
Where is your the pi.img file you are trying to shrink stored? You may have to specify a pathname to it (or I sometimes just change the directory I'm in to the directory with the .IMG file I'm shrinking and run PiShrink from there - then I don;t need to enter the pathname of the file.)

lar3ry, 2019-02-18:
Well, I probably have more clues now.
My first attempt was following the README.md, which means that I did not have a file called pi.img before issuing the pishrink.sh,

John-Mc, 2019-02-18:
You have to actually start with a **valid** .IMG file made from the SD card you wish to shrink, not just a file you make that ends in .img.

Probably the easiest way to do create an .IMG file on a Mac is with Apple Pi Baker, which will either burn a card from an image, or create an .img file from an SD card. 

On Windows, I know a couple of programs to burn an .IMG file to an SD card, but none that will go the other way (though I'm sure there are some out there).

After creating the .img file, you'll need to shrink that file while it is mounted on a linux system (such as your Pi running Raspian) using the pishrink command.

As an alternative, you can also make an .IMG file on your Pi of the SD Card from which you booted your Pi. One way to do this is with the command:

`sudo dd if=/dev/mmcblk0 of=<pathname>/mySD.img`

This copies the Pi's active boot SD Card to a new image named mySD.img (or whatever name you choose) at the <pathname> supplied ( without the <> marks). The pathname should be a formatted and mounted external drive attached to Pi's USB port). You **don't** want to store this .img file on your Pi's boot SD card. Note that this process can take a very LOOONG time (especially if your SD card is large), and your pi may appear to be locked up. If see that the green activity light on your Pi is still flashing, it's working. (On a 32 GB card, I start it up then walk away and do something else.) You can then run PiShrink against the resulting image.

**WARNING**: Someone who knows a whole lot more than me about Raspian/Linux has said that the above command could cause a problem with the data on the SD card. I've done it dozens of times without issue. It's recommended that you not be doing anything else on your Pi at the time. If someone else has a better/safer way of creating an IMG of an SD card on a Pi, I'd love to hear it (or if doing it as shown above is not a concern, I'd love to hear that as well).

lar3ry, 2019-02-19:
Ahh.. that makes sense. I was under the impression that *.img referred to in the instructions was going to be the output file, NOw that I know what's what, I see that a .img file with a different name can be made with the -s option.

I am trying the dd to make an image file now. Doing it from the Pi to a hard drive. Depending on how long it takes, I may find it easier to just grab the SD card and use Win32DiskImager. I was trying to avoid running VirtualBox on my Windows box to allow me to run pishrink, but I can always use SCP to send the image and do the pishrink on the Pi.

Looking forward to doing backups that can actually be reliably restored.

Thanks a LOT for your help. Much appreciated!

John-Mc, 2019-02-19:
You're close: the -s option controls whether the Operating system auto-expands on first boot to expand to fill whatever SD card it is installed on. If you just specify one name, PiShrink will look for that file shrink it and save it under the same name. If you specify the optional second name, PiShrink will create a new file under that name and leave the original .img file alone.

I could not recall if Win32DiskImager would make an .img file from an SD card (I know it can burn an .img to an SD card). If it will go the other way, than that is almost certainly faster and easier than doing it on the pi. (I then use the Pi to shrink that image, on a USB drive, since I've never gotten a virtual Linux machine set up on my Mac.)

lar3ry, 2019-02-19:
Yeah, I realized that when I went back and re-read the README. 

I still get errors, but I think it may be in the way I formatted the hard drive that I plugged into the Pi.
I also think I'll just bite the bullt and install VirtualBox and Debian.

Again, thanks for all your help!

John-Mc, 2019-02-19:
What format did you use for the Hard Drive? The Pi needs some help to read some formats.

lar3ry, 2019-02-20:
I used ext4, I think. I did not partition the drive before formatting. I assumed that if I did not patition it, it would have a single partition. Also, I did not supply a volume label. Here's what I get when I run pishrink

root@octopi:~# pishrink.sh ./tmp/dreamer.img
Error: Can't have a partition outside the disk!
tune2fs: Invalid argument while trying to open /dev/loop0
Couldn't find valid filesystem superblock.
mount: /dev/loop0: can't read superblock
md5sum: /tmp/tmp.PuFSaVT9Vv/etc/rc.local: No such file or directory
/usr/local/bin/pishrink.sh: line 78: [: !=: unary operator expected
umount: /tmp/tmp.PuFSaVT9Vv: not mounted
e2fsck: Invalid argument while trying to open /dev/loop0
/dev/loop0:
The superblock could not be read or does not describe a valid ext2/ext3/ext4
filesystem.  If the device is valid and it really contains an ext2/ext3/ext4
filesystem (and not swap or ufs or something else), then the superblock
is corrupt, and you might try running e2fsck with an alternate superblock:
    e2fsck -b 8193 <device>
 or
    e2fsck -b 32768 <device>

resize2fs 1.43.4 (31-Jan-2017)
resize2fs: Invalid argument while trying to open /dev/loop0
/usr/local/bin/pishrink.sh: line 151: [[: Couldn'tfindvalidfilesystemsuperblock.: syntax error: invalid arithmetic operator (error token is "'tfindvalidfilesystemsuperblock.")
/usr/local/bin/pishrink.sh: line 157: - Couldn'tfindvalidfilesystemsuperblock.: syntax error: invalid arithmetic operator (error token is "'tfindvalidfilesystemsuperblock.")
resize2fs 1.43.4 (31-Jan-2017)
resize2fs: Invalid argument while trying to open /dev/loop0
Couldn't find valid filesystem superblock.
ERROR: resize2fs failed...
mount: /dev/loop0: can't read superblock
mv: cannot stat '/tmp/tmp.PuFSaVT9Vv/etc/rc.local.bak': No such file or directory
umount: /tmp/tmp.PuFSaVT9Vv: not mounted
losetup: /dev/loop0: detach failed: No such device or address

lar3ry, 2019-02-20:
Just noticed that the Pi SD card uses vfat, so I partitioned the drive with 1 partition, made a vfat file system on it, did some test copies to and from it, and it all looks fine. I am now using Win32DiskImager to copy the SD card to a drive on my PC, and will be trying the pishrink on the Pi tonight.

Drewsif, 2019-02-20:
@John-Mc definitely don't mind you helping out!

Here are some YouTube videos that might help, they are related to retroarch but they will work on anything:
Taking an image of a pi on windows: https://www.youtube.com/watch?v=C1QdZSjc1C0
Virtualbox setup for pishrink: https://www.youtube.com/watch?v=E3oSdq4lyA0

closed 2019-02-21 by Drewsif

Morphy99, 2021-02-02:
I had this issue with an image I'd created of a SD card with Disks in Ubuntu. I created another image with Win32DiskImager in Windows in the same dir and it worked with PiShrink fine. The image I created in Ubuntu restores fine so not sure what's up with it.

RolfBly, 2023-06-14:
I'm having this issue with images created with Win32DiskImager in Window 10, running PiShrink in a docker container, as described [here](https://maskaravivek.medium.com/creating-and-shrinking-bootable-for-rasberry-pi-image-from-sd-card-d9355cb29ee8). The pishrink version in that docker container is 0.1.3. Or, that's what I get when I run

    docker run --privileged=true --rm --volume [myworkdir] mgomesborges/pishrink pishrink -v
    
I guess I'll have to run it on linux...

mgomesborges, 2023-06-14:
Hey @RolfBly, I believe that @John-Mc presents a good explanation of what the problem might be.

It is very likely that Win32DiskImager is creating an image that is not compatible with PyShrink. Have you tried using another tool to generate the image?

I know it's slow, but you could try `dd` for Windows for example. Take a look at this tutorial: [DD utility for Windows backup](https://serverspace.io/support/help/dd-utility-for-windows-backup/)

Or try to use Rufus, sounds like an interesting project: https://github.com/pbatard/rufus

RolfBly, 2023-06-14:
Hi @mgomesborges, I realised I had WSL with Ubuntu on the same machine, started that, downloaded & installed PiShrink (also version 0.1.3) as per the instructions in the [readme](https://github.com/Drewsif/PiShrink#readme) and it worked OK on the very same img file that wasn't recognised using the docker container. 

I just cloned the shrunken image to a different SD card, booted & everything was OK. 

One thing I noticed: the command line in the [docker repo](https://hub.docker.com/r/mgomesborges/pishrink) switches `-pZv`, but the -p switch wasn't recognised. That command is almost literally copied in the tutorial I started out with, so maybe the -p switch used to do something, and was removed in a later version? Just a guess.

mgomesborges, 2023-06-14:
@RolfBly thanks for your feedback! You're right, the option `-p` was used to remove logs, apt archives, dhcp leases and ssh hostkeys, which was removed in #257. I'll update the code and DockerHub soon.

Anyway, I'm glad you managed to find a solution.