Drewsif/PiShrink #97: Errors in Ubuntu 14.04.5 LTS

jondkelley, 2019-04-07:
```
Copying img.img to new.img...
/dev/loop0 has unsupported feature(s): metadata_csum
e2fsck: Get a newer version of e2fsck!
resize2fs 1.42.9 (4-Feb-2014)
resize2fs: Invalid argument while trying to open /dev/loop0
/x.sh: line 151: [[: Couldn'tfindvalidfilesystemsuperblock.: syntax error: invalid arithmetic operator (error token is "'tfindvalidfilesystemsuperblock.")
/x.sh: line 157: 455680 - Couldn'tfindvalidfilesystemsuperblock.: syntax error: invalid arithmetic operator (error token is "'tfindvalidfilesystemsuperblock.")
resize2fs 1.42.9 (4-Feb-2014)
resize2fs: Invalid argument while trying to open /dev/loop0
Couldn't find valid filesystem superblock.
ERROR: resize2fs failed...
mount: /dev/loop0: can't read superblock
mv: cannot stat ‘/tmp/tmp.MgvuY4qbxK/etc/rc.local.bak’: No such file or directory
umount: /tmp/tmp.MgvuY4qbxK: not mounted
```

JustMeErazem, 2019-04-07:
In the readme of the project it states that if using Ubuntu you will probably get an e2fsck error, which I'm guessing is due to Ubuntu having horribly outdated software.

The readme recommends using Ubuntu 16.04 and up, so besides manually building e2fsck this is probably your only option.

jondkelley, 2019-04-07:
Okay awesome. I'll try the latest Centos 7 image.

For the record 16.04 was also problematic (just finished trying that) for the same reasons. Presumably because e2fsck is too old in Ubuntu.

jondkelley, 2019-04-07:
Seems like 16.10 is the requirement, so I'll try that before Centos

JustMeErazem, 2019-04-07:
Try that and report your findings. Also the development of this project seems to have stopped, so you should probably check if there has been a fork, since there are a few useful pull requests not being merged.

Disclaimer, this is self-promotion, but I have one such fork [over at Gitlab](https://gitlab.com/JustMeErazem/pishrinked) with the added bonus of using rsync to copy the img (verbose output) and one of the pull requests that compresses the output iso. Check it out if you want.

framps, 2019-04-07:
>  Also the development of this project seems to have stopped, 
@JustMeErazem  Would be interesting to know what's the source of this conclusion

jondkelley, 2019-04-07:
I jumped to Ubuntu 18.04.1 LTS \n \l and got the script working.
I need to work on my image a bit more though 

```
sudo /tmp/pismall.sh img.img new.img
Copying img.img to new.img...
Creating new /etc/rc.local
pi-star: 49822/114016 files (0.2% non-contiguous), 418730/455680 blocks
resize2fs 1.44.1 (24-Mar-2018)
ERROR: Image already shrunk to smallest size
```

Otherwise this should probably work.

For those interested here is my Vagrantfile

```
# -*- mode: ruby -*-
# vi: set ft=ruby :

Vagrant.configure("2") do |config|
	config.vm.hostname = "pishrinker"

        config.vm.box = "bento/ubuntu-18.04" 
	config.vm.network :forwarded_port, guest: 5672, host: 55672
	config.vm.network :forwarded_port, guest: 15672, host: 55672

config.vm.provision "file", source: "Apr2019.img", destination: "img.img"
config.vm.provision "file", source: "pishrink.sh", destination: "/usr/bin/pishink.sh"
end
```

closed 2019-04-07 by jondkelley

JustMeErazem, 2019-04-07:
@framps The latest update to the script has been a year ago, the pull requests have not been answered or closed and the only recent update has been a fix in a typo in the readme. It might be that the dev is busy or whatever, it's just my opinion that he has stopped the development on this project.

But that doesn't mean that the project is outdated now, of course it isn't, it works great and does what is says perfectly.

framps, 2019-04-07:
@JustMeErazem Last commit is from1/17/2019 ...

JustMeErazem, 2019-04-08:
@framps Yes, I said that "the only recent update has been a fix in a typo in the readme".

The last update to the script has been about a year ago.