Drewsif/PiShrink #99: Boot hang after "random: crng init done"

Laquio, 2019-04-15:
I did try creating a backup image using dd command, hoping to shrink it a custom image that is bootable ready.
I follow this link: https://medium.com/platformer-blog/creating-a-custom-raspbian-os-image-for-production-3fcb43ff3630

Steps I took:
1. RPI is boot via usb (16gb) with raspbian stretch.
2. Insert my working sdcard (4gb with custom raspbian OS and packages).
3. Create a back up image of my 4gb sdcard using dd command save it to "/home/pi/" dir
4. execute the PiShrink.sh (was successful).
5. burn the shrunk image via Etcher.
6. It boots but failed to proceed after 5 seconds stops with an error message "random: crng init done".

Please advice.

Drewsif, 2019-07-24:
Can you post a console log of taking the image? Thats might help

chtoller, 2019-12-31:
Hi there,

having the same issue right now.
While I can restore the original image to a different SD card and that boots just fine, restoring the smaller image created by PiShrink fails to boot.
Boot process is stuck at "random: crng init done".

Here's the console log of creating the image and running PiShrink:
`christian@debian:~$ sudo dd bs=4M if=/dev/sdc of=skyvpn-client-orig.img
[sudo] password for christian: 
3798+1 records in
3798+1 records out
15931539456 bytes (16 GB, 15 GiB) copied, 282.374 s, 56.4 MB/s
christian@debian:~$ sudo pishrink.sh skyvpn-client-orig.img skyvpn-client.img
pishrink.sh v0.1.2
pishrink.sh: Copying skyvpn-client-orig.img to skyvpn-client.img......
pishrink.sh: Gatherin data...
Creating new /etc/rc.local
pishrink.sh: Checking filesystem...
rootfs: 41296/489472 files (0.1% non-contiguous), 315464/2097152 blocks
resize2fs 1.43.4 (31-Jan-2017)
pishrink.sh: Shrinking filesystem...
resize2fs 1.43.4 (31-Jan-2017)
Resizing the filesystem on /dev/loop0 to 369418 (4k) blocks.
Begin pass 2 (max = 1476)
Relocating blocks             XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
Begin pass 3 (max = 64)
Scanning inode table          XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
Begin pass 4 (max = 4325)
Updating inode references     XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
The filesystem on /dev/loop0 is now 369418 (4k) blocks long.

pishrink.sh: Shrinking image...
pishrink.sh: Shrunk skyvpn-client.img from 15G to 1.5G...
`

Tried different SD cards, same behavior.

Any ideas?

Drewsif, 2020-06-17:
It looks like its a known problem. See more info here https://www.raspberrypi.org/forums/viewtopic.php?t=209093

closed 2020-06-17 by Drewsif