Drewsif/PiShrink #100: Easy 14% improvement in zipped size

hraftery, 2019-05-23:
For those that are zipping their shrunk images (and I suggest you do - I routinely get 70% deflation and Etcher happily reads zips), there's a quick fix to get an easy extra ~17% reduction in file size: simply write zeros to all the empty space on the shrunk ext4 partition.

```
$ diff pishrink-orig.sh pishrink.sh 
38c38
< for command in parted losetup tune2fs md5sum e2fsck resize2fs; do
---
> for command in parted losetup tune2fs md5sum e2fsck resize2fs dcfldd; do
186a187,192
> 
> #Zero-fill the free space for better compression
> mount "$loopback" "$mountdir"
> dcfldd if=/dev/zero of="$mountdir"/zero.txt
> rm "$mountdir"/zero.txt
> umount "$mountdir"
```

kapitainsky, 2019-05-23:
Good tip! From my side I would also suggest to use xz commpressor - images deflation will be significantly higher than with zip (and Etcher also supports xz compressed images out of the box)

hraftery, 2019-05-24:
Wow. Percentage deflation stats on two images I just tried:

```
|     | img1  | img2  |
| zip | 71.37 | 70.38 |
| xz  | 80.27 | 78.66 |
```

Works out to be ~250MB instead of ~410MB from a ~1.5GB image. But damn, it takes some time:

```
|     |  img1  |  img2  |
| zip |  1m50s |  2m20s |
| xz  | 16m45s | 20m57s |
```

This was on Ubuntu Server running on a HDD, operating on local files of 1.5GB and 1.9GB, using `time zip file.img` and `time xz -k file.img`.

Anyway, here's the original diff in unified format because that might be easier to use as a patch.
```
$ diff -u pishrink-orig.sh pishrink.sh
--- pishrink-orig.sh	2019-05-23 06:58:13.000000000 +0000
+++ pishrink.sh	2019-05-23 08:31:36.000000000 +0000
@@ -35,7 +35,7 @@
 fi
 
 #Check that what we need is installed
-for command in parted losetup tune2fs md5sum e2fsck resize2fs; do
+for command in parted losetup tune2fs md5sum e2fsck resize2fs dcfldd; do
   which $command 2>&1 >/dev/null
   if (( $? != 0 )); then
     echo "ERROR: $command is not installed."
@@ -185,4 +185,10 @@
 truncate -s $endresult "$img"
 aftersize=$(ls -lh "$img" | cut -d ' ' -f 5)
 
+#Zero-fill the free space for better compression
+mount "$loopback" "$mountdir"
+dcfldd if=/dev/zero of="$mountdir"/zero.txt
+rm "$mountdir"/zero.txt
+umount "$mountdir"
+
 echo "Shrunk $img from $beforesize to $aftersize"
```

kapitainsky, 2019-05-24:
xz is slower for compression but not much for decompression. If you are happy to sacrifice a bit of compression ratio for speed (like 1% maybe) try pxz (this is what I use). It uses multiple cpu cores increasing compression speed multiple times. Also you can change xz default compression level - with the lowest one it is still a bit better than gzip and in this case also faster.

kapitainsky, 2019-05-24:
and one questions - any reason why you use dcfldd instead of dd? for most people it means requirement to install it.

JustMeErazem, 2019-05-24:
@kapitainsky Some implementations of dd don't show progress (even with the "status=progress" flag), but dcfldd does.

hraftery, 2019-05-24:
On the use of `dcfldd`, I lifted this out of a script I was developing which included the extraction of the image from an SD Card. That long process warrants installation of the progress indicating `dcfldd`, but in this case it probably doesn't. So I should have used `dd`.

Drewsif, 2019-07-24:
Ill look into adding it, thanks for the suggestion 😄

closed 2019-07-24 by Drewsif

rsaxvc, 2021-05-17:
zerofree will do just the slack space for you

rsaxvc, 2021-05-17:
Related to #59