Drewsif/PiShrink #101: Unexpected Inconsistency

modem7, 2019-05-25:
I'm backing up with clonepi, standard raspbian install, but getting the following error: 

```
pi@NginxPi:~ $ sudo pishrink.sh /mnt/pibackup/incremental/nginx-pi.img nginx-pi-                                                                                                                                                                  shrunk.img
Copying /mnt/pibackup/incremental/nginx-pi.img to nginx-pi-shrunk.img...
Creating new /etc/rc.local
rootfs: Entry 'info' in /var/webmin/modules/system-status (386783) has deleted/u                                                                                                                                                                  nused inode 381038.  CLEARED.
rootfs: Unattached inode 381034

rootfs: UNEXPECTED INCONSISTENCY; RUN fsck MANUALLY.
        (i.e., without -a or -p options)
resize2fs 1.43.4 (31-Jan-2017)
resize2fs: Invalid argument while trying to open /dev/loop0
/usr/local/bin/pishrink.sh: line 151: [[: Couldn'tfindvalidfilesystemsuperblock.                                                                                                                                                                  : syntax error: invalid arithmetic operator (error token is "'tfindvalidfilesyst                                                                                                                                                                  emsuperblock.")
/usr/local/bin/pishrink.sh: line 157: 7779456 - Couldn'tfindvalidfilesystemsuper                                                                                                                                                                  block.: syntax error: invalid arithmetic operator (error token is "'tfindvalidfi                                                                                                                                                                  lesystemsuperblock.")
resize2fs 1.43.4 (31-Jan-2017)
resize2fs: Invalid argument while trying to open /dev/loop0
Couldn't find valid filesystem superblock.
ERROR: resize2fs failed...
mount: /dev/loop0: can't read superblock
mv: cannot stat '/tmp/tmp.PzeB80S9HT/etc/rc.local.bak': No such file or director                                                                                                                                                                  y
umount: /tmp/tmp.PzeB80S9HT: not mounted
losetup: /dev/loop0: detach failed: No such device or address
```

```
Disk /dev/mmcblk0: 29.7 GiB, 31914983424 bytes, 62333952 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x9bb4c1ed

Device         Boot Start      End  Sectors  Size Id Type
/dev/mmcblk0p1       8192    98045    89854 43.9M  c W95 FAT32 (LBA)
/dev/mmcblk0p2      98304 62333951 62235648 29.7G 83 Linux

```

Not quite sure where to go from here. 

Thanks!

kapitainsky, 2019-05-25:
mount image manually:

`sudo losetup -P /mnt/pibackup/incremental/nginx-pi.img`

check which loop device it is attached to

`sudo losetup`

run filesystem scan/repair (if loop0 used)

`sudo e2fsck -fy /dev/loop0p2`

detach loop device

`sudo losetup -d loop0`

kapitainsky, 2019-05-25:
File system inconsistencies can happen quite easy when you image live system. I would suggest you at least stop non essential services during image capture.  Or create image offline.

modem7, 2019-05-26:
Heya, 

Thanks for the above!

When I try the first command, I get:

```
pi@NginxPi:~ $ sudo losetup -P /mnt/pibackup/full/nginx-pi-shrunk.img
losetup: /mnt/pibackup/full/nginx-pi-shrunk.img: failed to use device: No such device
```

```
pi@NginxPi:~ $ sudo losetup -f
/dev/loop0
```

kapitainsky, 2019-05-26:
well. It works for me. Something with the your setup. Google is your friend:)

kapitainsky, 2019-05-26:
And when you mount it have you tried to run file system check?

bggardner, 2019-07-01:
I had the same problem.  Running `sudo e2fsck -f /dev/loop0` (without `-p`) and saying "yes" to fixing the "corrupted orphan link list" allowed `pishrink` to complete successfully.  My image was generated via `dd`.

Drewsif, 2020-06-17:
Closing since it looks like a solution was posted

closed 2020-06-17 by Drewsif