Drewsif/PiShrink #104: pishrink hangs and never finishes

LuckyGinger, 2019-07-22:
Getting an error in Linux Mint 19.1 running in virtual box as well as Ubuntu 18.04.2

PiShrink hangs and never finishes when trying to shrink an ARMBIAN 5.75 img of an OrangePi. 

> XXXX@XXXX-VirtualBox:/media/sf_D_DRIVE/images$ sudo pishrink.sh new.img
> Creating new /etc/rc.local

Leaving it overnight and doesn't finish. I've seen people around on forums suggest using PiShrink for OrangePi images. Seems this is not the case?

Drewsif, 2019-07-24:
Are you using shared folders on virtual box?

LuckyGinger, 2019-07-24:
So I've tried it a couple ways. Using shared folders as you can see in the path above `sf_D_DRIVE`. That's when it hangs for eternity.

The following error happens when I try to do everything within the VM. This is the error I get inside the Linux Mint 19.1 VM.

> XXXX@XXXX-VirtualBox:~/Images$ sudo pishrink.sh imgClone.img new.img
> Copying imgClone.img to new.img...
> Creating new /etc/rc.local
> /dev/loop0: Inodes that were part of a corrupted orphan linked list found.  
> 
> /dev/loop0: UNEXPECTED INCONSISTENCY; RUN fsck MANUALLY.
> 	(i.e., without -a or -p options)
> resize2fs 1.44.1 (24-Mar-2018)
> resize2fs: Invalid argument while trying to open /dev/loop0
> /usr/local/bin/pishrink.sh: line 151: [[: Couldn'tfindvalidfilesystemsuperblock.: syntax error: invalid arithmetic operator (error token is "'tfindvalidfilesystemsuperblock.")
> /usr/local/bin/pishrink.sh: line 157: 7477452 - Couldn'tfindvalidfilesystemsuperblock.: syntax error: invalid arithmetic operator (error token is "'tfindvalidfilesystemsuperblock.")
> resize2fs 1.44.1 (24-Mar-2018)
> resize2fs: Invalid argument while trying to open /dev/loop0
> Couldn't find valid filesystem superblock.
> ERROR: resize2fs failed...
> mount: /tmp/tmp.3bFpXAf7Wy: can't read superblock on /dev/loop0.
> mv: cannot stat '/tmp/tmp.3bFpXAf7Wy/etc/rc.local.bak': No such file or directory
> umount: /tmp/tmp.3bFpXAf7Wy: not mounted.
> losetup: /dev/loop0: detach failed: No such device or address

Hope this helps

Drewsif, 2020-06-17:
There were some changes made that should fix this issue hopefully. If not feel free to re-open!

closed 2020-06-17 by Drewsif