Drewsif/PiShrink #106: dow not work on Linux debian 4.19.0-5-686-pae #1 SMP Debian 4.19.37-5+deb10u1 (2019-07-19) i686 GNU/Linux

rudiratlos, 2019-08-10:
pishrink.sh ./pumpmaster_16GB.img ./pumpmaster.img
pishrink.sh v0.1
Copying ./pumpmaster_16GB.img to ./pumpmaster.img......
Gatherin data...
Error: /data/home/sfischer/projects/pas/pump/publish/pumpmaster.img: unrecognised disk label
losetup: ./pumpmaster.img: Warning: file does not fit into a 512-byte sector; the end of the file will be ignored.
tune2fs: Invalid argument while trying to open /dev/loop0
Couldn't find valid filesystem superblock.
mount: /tmp/tmp.swt2Epnuy5: wrong fs type, bad option, bad superblock on /dev/loop0, missing codepage or helper program, or other error.
md5sum: /tmp/tmp.swt2Epnuy5/etc/rc.local: No such file or directory
/usr/local/bin/pishrink.sh: line 126: [: !=: unary operator expected
umount: /tmp/tmp.swt2Epnuy5: not mounted.
Checking filesystem...
e2fsck: Invalid argument while trying to open /dev/loop0
/dev/loop0: 
The superblock could not be read or does not describe a valid ext2/ext3/ext4
filesystem.  If the device is valid and it really contains an ext2/ext3/ext4
filesystem (and not swap or ufs or something else), then the superblock
is corrupt, and you might try running e2fsck with an alternate superblock:
    e2fsck -b 8193 <device>
 or
    e2fsck -b 32768 <device>

ERROR occured in line 200: fsck failed with rc 0

rudiratlos, 2019-08-12:
solved it: just had to put PATH variable in my calling script:
PATH=$PATH:/sbin:/usr/bin:/usr/sbin:/usr/local/sbin; export PATH

in line 86, check of dmidecode is missing

framps, 2019-08-23:
Great you found the root cause for your issue.

> in line 86, check of dmidecode is missing

Not sure I get your point. dmidecode is not used by pishrink :wink: