Drewsif/PiShrink #107: MacOS: ERROR occured in line 86: parted is not installed.

georgbachmann, 2019-08-20:
I'd like to shrink my pi image to put it on a smaller SD card, but i get the following error: 

```ERROR occured in line 86: parted is not installed.```

I tried installing `parted` using homebrew, but can't seem to find it :(
Any idea on how to use the script on MacOS?

framps, 2019-08-20:
> I tried installing `parted` using homebrew, but can't seem to find it :(

I have no clue about MacOS. But as far as I know there is some kind of Unix used under the hood. Open a commandline window and make sure `which parted` returns a fully qualified file name. Then pishrink will be able to find the missing tool.

kapitainsky, 2019-08-21:
pishrink does not work on macOS. there is more different things missing than parted. 

The solution is to run it from within linux VM

georgbachmann, 2019-08-21:
@kapitainsky ok... cool... will do that... thanks!

closed 2019-08-21 by georgbachmann

3Dietrich, 2021-02-10:
I also fell into the trap and was now looking for "parted" for mac... 
Please mention ("NOT FOR MAC" or something like that) in the description! Or you leave behind (frustrated) Mac users!

mgomesborges, 2021-02-10:
Hey @georgbachmann @framps @kapitainsky @3Dietrich  you can use [PiShrink dockerized](https://github.com/mgomesborges/pishrink-docker) | [Docker Hub PiShrink](https://hub.docker.com/r/mgomesborges/pishrink):

## macOS Setup

1. [Install Docker Desktop on Mac](https://docs.docker.com/docker-for-mac/install/)

2. Make a copy of a Raspberry Pi SD card that you want to shrink ([see instructions here](https://github.com/mgomesborges/raspberry-pi/blob/master/setup/clone-sd-card.md)).

3. Using the Terminal, access the directory containing the Raspberry Pi image:

```bash
cd ~/Directory-with-RPi-image
```

4. Run PiShrink dockerized:

```bash
docker run --privileged=true --rm \
    --volume $(pwd):/workdir \
    mgomesborges/pishrink \
    pishrink -pZv IMAGE_NAME.img NEW-IMAGE_NAME.img
```

3Dietrich, 2021-02-10:
@mgomesborges WOW!
I tried, it worked (at least it ran through and left me a 663MB file "image.img.xz")
thank you! 
the [link to your project](https://github.com/mgomesborges/raspberry-pi/blob/master/setup/clone-sd-card.md#clone-and-shrink-a-raspberry-pi-image) should also be in the readme!!

lizziemac, 2024-05-15:
just a 2024 update for anyone else who stumbles upon the gem in this issue, the docker container still works!  some small things:
1. Permissions
    a. I personally had to set the file permissions to `666` with `sudo chmod 777 <image>`
    b. I also needed to make sure that I had the correct ownership. For me I was able to proceed with `sudo chown $(whoami):staff <image>`
2. I also had to explicitly use the `v0.1.4` release for Docker image
3. I'm not sure what the `-p` argument was for, and it no longer exists, so I removed it and it still worked

Drewsif, 2024-10-05:
If anyone finds this later, new instructions in the README