Drewsif/PiShrink #110: PiShrink does nothing?

balouken, 2019-10-22:
Hi,

I'm try'n to get PiShrink to work.
But when i enter the code for execute the script, the script is gatherin data for a whole day, and nothing happens. The only thing i see, is that there is a disk mounted.
Can anybody help me out?

balouken@balouken-VirtualBox:/media/sf_PiShrink-master/pi$ sudo chmod +x pishrink.sh
[sudo] password for balouken: 
balouken@balouken-VirtualBox:/media/sf_PiShrink-master/pi$ sudo ./pishrink.sh RegaladV4.img
pishrink.sh v0.1.2
pishrink.sh: Gatherin data...

framps, 2019-10-22:
pishrink writes a log file called pishrink.log. Please provide the contents of this log file.

balouken, 2019-10-22:
> pishrink writes a log file called pishrink.log. Please provide the contents of this log file.

where can i find this log in Ubuntu? Can't find it.

framps, 2019-10-22:
Should be in your current directory.

balouken, 2019-10-22:
no, there is nog logfile in the same directory of the script?

framps, 2019-10-22:
If you are in directory /home/balouken and call pishrink with sudo ./pishrink.sh there should be a file called pishrink.log in your current directory which is /home/blouken.

In other words: `ls pishrink.*` should list the log file.

balouken, 2019-10-22:
No log file to find

balouken@balouken-VirtualBox:/media/sf_PiShrink-master$ ls
**howto.txt  LICENSE  pi  README.md**
balouken@balouken-VirtualBox:/media/sf_PiShrink-master$ cd pi
balouken@balouken-VirtualBox:/media/sf_PiShrink-master/pi$ ls
**pishrink.sh  RegaladV4.img**
balouken@balouken-VirtualBox:/media/sf_PiShrink-master/pi$ 
balouken@balouken-VirtualBox:/media/sf_PiShrink-master/pi$ 
balouken@balouken-VirtualBox:/media/sf_PiShrink-master/pi$ ls pishrink.*
**pishrink.sh**

framps, 2019-10-22:
Hm ... that's weird. Let's try another approach: 
`sudo bash -x ./pishrink.sh`
This will create a lot of messages on your screen. Please show the last 10 lines.

balouken, 2019-10-22:
I give you more, cause the last 10 lines are sayin' nothing i think :-)

[image]

framps, 2019-10-22:
Sorry. My fault. Please use
`sudo bash -x ./pishrink.sh RegaladV4.img`

balouken, 2019-10-22:
No problem, you're try' to help me :)

Journal backup:           inode blocks'
++ grep '^Block size:'
++ tr -d ' '
++ cut -d : -f 2
+ blocksize=4096
+ logVariables 167 tune2fs_output currentsize blocksize
+ '[' false = true ']'
+ '[' false = false ']'
++ mktemp -d
+ mountdir=/tmp/tmp.cLONc6lMVB
+ mount /dev/loop9 /tmp/tmp.cLONc6lMVB
++ cut -d ' ' -f 1
++ md5sum /tmp/tmp.cLONc6lMVB/etc/rc.local
+ '[' 0542054e9ff2d2e0507ea1ffe7d4fc87 '!=' 0542054e9ff2d2e0507ea1ffe7d4fc87 ']'
+ umount /tmp/tmp.cLONc6lMVB

framps, 2019-10-22:
`umount /tmp/tmp.cLONc6lMVB` is the last command executed by pishrink and hangs for some unknown reasons.

I saw you're running pishrink in VB and try to shrink an image mounted on /media/sf_PiShrink-master.

I frankly don't understand why an umount on /tmp hangs in your environment :disappointed: I think it's somehow related to the fact you're running VB. I run native Linux and thus have no experience with VB and pishrink :cry:

kapitainsky, 2019-10-22:
pishrink won’t work on VB box shares. Like on any other network shares. Put your image on ext4 filesystem

kapitainsky, 2019-10-22:
VB provides shares via emulated network drive.

kapitainsky, 2019-10-22:
I use it myself in VMware but it requires ext4 filesystem mounted. Create new disk in VB and mount it properly. Then all will work

kapitainsky, 2019-10-22:
Shares are more like NFS filesystem. Good to copy files between. But they don’t provide full local filesystem features.

balouken, 2019-10-22:
there are a lot of youtubers where it works with VB en shared folder...
New disk in VM without running the VM i supose
I give it a try...

framps, 2019-10-22:
> pishrink won’t work on VB box shares. Like on any other network shares. 
I would appreciate if you have some more details on this. Any links handy?

kapitainsky, 2019-10-22:
You can keep trying. It won’t work. Please use google. There is massive difference between local and network filesystem. Pishrink is very well done and clever tool I use myself. But it will never work on images stored on network drives.

kapitainsky, 2019-10-22:
Just move image to locally mounted ext4 filesystem. Then all will work.

balouken, 2019-10-22:
ok, script works when i locate new drive.
But it can not be shrinked... damn!

[image]

Drewsif, 2020-06-17:
Ive added a warning about shared folders now to the README

closed 2020-06-17 by Drewsif