Drewsif/PiShrink #114: Predefined expand size possible?

metbl09, 2019-10-31:
As far as I can see, there's two options regarding expanding. Expand to max size of sd card or don't expand at all.
Is it possible to choose a size to which the image expands? E.g. Expand a 0.xGB image to 2GB although it's a 8GB sd card.

John-Mc, 2019-11-30:
It it is possible, it would need to check that the card is at least as big as the specified size.

Drewsif, 2020-06-17:
It's possible, but would be a pain to do. I'm not clear on why you would want to do it. Whats the thought process behind only partly expanding?

PK232, 2022-05-11:
I work with images frequently and Pishrink certainly shortens the time to write an image to a card.  As for creating images from a card, it appears the only way shorten the read time is to use a card with a higher speed rating.  Unfortunately it seems that the higher speed cards are more and more only available in the larger sizes and using those cards does not speed up the image creation time.  If you buy a card with twice the speed rating and it is twice as big as what you currently use, the net gain close to zero.  At present, based on what I have read, you have only have two choices when shrinking the image which are either full expansion on boot or no expansion on boot.   It would be nice if you could set the expansion size to say 14 GB.   It would allow you plenty of space to work with but shorten the read time creating an image if it was a 64 GB card.

Drewsif, 2023-02-07:
I don't see a huge portion of the community looking for this feature so I'm going to close this issue until the time it gets more traction. It would be a bit of work and does not feel worth the development time. Sorry 😞

closed 2023-02-07 by Drewsif