Drewsif/PiShrink #116: parted failed with rc 0

jordipalet, 2019-11-18:
It looks like everything is going fine, but at the end I get:

Line 287
partnewsize: 11562852352
newpartend: 13550952448
Error: Can't have overlapping partitions.
pishrink.sh: ERROR occured in line 296: parted failed with rc 0

Any hint?

situssoft, 2020-02-06:
Me too, any idea how to fix this?

situssoft, 2020-02-06:
Incase anyone else ends up here, for my case of shrinking a raspbian lite based image the script is attempting to create the wrong type of partition on line 307. For my setup the partition type is logical, by changing this line from primary to logical the script completed without error.

framps, 2020-02-06:
> Incase anyone else ends up here, for my case of shrinking a raspbian lite based image the script is attempting to create the wrong type of partition on line 307. 

Thank you very much for your update.

> For my setup the partition type is logical, by changing this line from primary to logical the script completed without error.

What's special on your setup? If we have to handle this in the code we have to know when to go left and when to go right :wink:

situssoft, 2020-02-06:
Ok, time to be humble. After further investigation I have realised that this problem is because I had an image createed from NOOBS. The partitioning is entirly different and I did read that this isn't supported yet.

So for clarity. You will get this error if you attempt to shrink an image that is the result of a NOOBS install.

To avoid that, install from the raspbian image (I'm using raspbain lite) and all will be good.

lurch, 2020-04-14:
Duplicate of #44 ?

Drewsif, 2020-06-13:
This is a duplicate of the other NOOBS related issues

closed 2020-06-13 by Drewsif