Drewsif/PiShrink #117: use pigz, for multicore crunshing

ThorstenS-linux, 2019-11-30:
The debian package pigz contains a zip variant which uses all CPUs when shrinking. This speeds things up a lot on today's multi-core machines.
I suggest to test if pigz is present and then use it instead of gzip.

thanks a lot for this script!
/thorsten

OmegaSquad82, 2020-01-11:
Hi, it seems there are better compression algorithms available for quite some time now. This [benchmark](https://community.centminmod.com/threads/round-3-compression-comparison-benchmarks-zstd-vs-brotli-vs-pigz-vs-bzip2-vs-xz-etc.17259/) shows that xz saves more disk space.

- Comparing xz vs. gzip the compression speed is ~90% slower but has a ~41,17% better ratio.
- Using pxz vs. pigz is ~93% slower but still has ~39,77% better ratio.

Personally I'd prefer choice between using pigz and xz so leave caller the choice to trade some time for space.

framps, 2020-01-11:
I created a [branch in my fork](https://github.com/framps/PiShrink/tree/other_zip_tools) which allows to use pgz, xz or pxz instead of gzip. Give it a try :wink: 

**DISCLAIMER**: I just added the code to support the additional zip tools and tested whether the correct call of the zip tool will be executed by pishrink. I never executed the code to test whether zipping with these tools works. This would have increased the development time significantly which is too much I'm willing to spend for adding this additional zip tool support. 

If you find any issue just let me know. I'll fix it if there is no test execution required from my side :smirk:

OmegaSquad82, 2020-01-11:
It seems that pgz is not developed anymore for several years now, but xz got multithreading along the way.

framps, 2020-01-12:
I created following list to track tests of my [fork](https://github.com/framps/PiShrink/tree/other_zip_tools) which supports other zip tools. In order to spread the test effort everybody who tested one zip tool with pishrink is kindly asked to report the test result and I'll update the list. If everything is tested successfully I'll create a PR and ask @Drewsif to merge the code into master.

 - [ ] - gzip tested
 - [ ] - pigz tested
 - [ ] - xz tested
 - [ ] - pxz tested

The code is flexible enough to add other zip tools very quickly. If there is need for other zip tools just let me know the tool name and the options to use :wink:

framps, 2020-01-12:
> It seems that pgz is not developed anymore for several years now

My fault. My fork supports pigz and not pgz. I fixed [my typo](https://github.com/Drewsif/PiShrink/issues/117#issuecomment-573322911)

OmegaSquad82, 2020-01-20:
@framps I'm going to test some tonight, thx for your time! ;c)

framps, 2020-01-20:
@OmegaSquad82 Great :+1: Just let me know if you find any issues.

OmegaSquad82, 2020-01-22:
@framps I used unmodified uncompressed Raspbian image obtained from https://downloads.raspberrypi.org/raspbian_lite_latest with size ~2,1 GiByte and sha256 fef882aa980cda898421e0f71328033c02cc314ed438b5ec8f39b03e641f31b1
`$ for tool in gzip pigz xz pxz; do sudo pishrink -zi $tool 2019-09-26-raspbian-buster-lite.img shrinked-$tool.img; done`

Which produced an intermediate, uncompressed image of 1,7G each and then
`pishrink: Shrunk shrinked-gzip.img.gz from 2,1G to 431M...
pishrink: Shrunk shrinked-pigz.img.gz from 2,1G to 431M...
pishrink: Shrunk shrinked-xz.img.xz from 2,1G to 273M...
pishrink: Shrunk shrinked-pxz.img.xz from 2,1G to 280M...`

From my pov it does not make much sense to integrate pxz as xz already employs multithreading when used in conjunction with -T0 option. It then runs as many threads as logical cores available. Adding that pxz is buggy and not actively maintained atm. It also crashed because of OOM on my 4GB laptop so I had to exit Xorg and stop some services on top of that.

Installed software versions
`gzip 1.6-5ubuntu1
pigz 2.4-1
xz (XZ Utils) 5.2.2 (liblzma 5.2.2)
pxz 4.999.99~beta5+gitfcfea93-2 (liblzma 5.2.2)`

framps, 2020-01-22:
@OmegaSquad82 Thank you very much for your time you spent for testing my branch. It's interesting to see the different sizes of the compressed images. You didn't provide any information about the compression time but I expect pigz to be faster than gzip :wink: 

> From my pov it does not make much sense to integrate pxz as xz already employs multithreading when used in conjunction with -T0 option

I didn't spend any time on evaluating whether the different tools are well supported. What's your opinion on pxz and xz? Should we drop pxz and use option -T0 all the time for xz? An alternative would be to allow gzip and xz an an option for -i only and to have an additional option to define whether parallel zip processing should be used (-p is already used, maybe -a from p`a`rallel would be the right option for this?)

OmegaSquad82, 2020-01-23:
@framps you're right and xz takes considerably longer and especially the memory footprint with -9 is unbearable but obviously it is a tradeoff. The algorithm (lzma2) used by (p)xz does not allow parallel execution so both tools divide the input stream into 1 separate for each thread which results in slightly more overhead but much faster execution.

As using 1/n core is inefficient and the increased file size is not that bad given the overall ratio of xz vs. gzip I'd drop pxz and always call `xz -T0`. Given that standard compression (`-6`) conserves on memory and should perform well enough even on a RasPi. Personally I'd also use `-v` to inform the user that there actually is something going on. What do you think? ,c)

On my TravelMate B with Celeron© CPU  N3160  @ 1.60GHz × 4 and 4 GB
`$ time xz -kv -T0 -9 shrinked.img` (-k retains original file) => mem used ~4GB so again OOM error
`shrinked.img (1/1)
  100 %     279,5 MiB / 1.690,9 MiB = 0,165   3,1 MiB/s       9:11`

vs.

`$ time xz -kv -T0 shrinked.img` (implies -6) => mem used <<1GB which really helps a lot
`shrinked.img (1/1)
  100 %     301,3 MiB / 1.690,9 MiB = 0,178   4,4 MiB/s       6:28`

framps, 2020-01-23:
@OmegaSquad82 Thank you very much for testing the different options. I think we should change the code as follows:
1. option -i accepts gzip and xz only (no pigz and pxz)
2. option -a turns on parallel processing and uses pigz with option -9 for -i gzip and xz with option -T0  for -i xz
3. option -v turns on verbose mode of the compression tools

What's your opinion on my proposal?

OmegaSquad82, 2020-01-23:
@framps thank you, this is good, I love that it avoids breaking changes, I'm content. ;c)
@ThorstenS-linux are you satisfied with the proposal adding new options?

OmegaSquad82, 2020-02-03:
@framps I'd really like what you've proposed. ;c)

framps, 2020-02-03:
@Drewsif Does it make sense for you to add this feature to pishrink?

Drewsif, 2020-02-07:
I'm actually fine defaulting to trying to use pigz over gzip if present on the machine. I think a lot of users may not understand why they want multicore but them seeing the time savings on compression would be nice. If we add an option to limit it I would want to make it a core count option for more direct control.

Along the same lines, I'm not a huge fan of the proposed command line for that reason. I'm inclined to copy tar's command line and add `-j` for bzip2 and `-J` for xz. Or, since I'm not a big bzip2 person and don't care if it makes it in, making `-Z` use xz. I think its just a bit cleaner overall, but of course I'm open to feedback

framps, 2020-02-07:
I just pushed a version of pishrink in my branch which doesn't use option -i <ziptool> any more but hast option -z for gzip and -Z for xz. -a enables parallel compression. -v enables verbose compression.

@OmegaSquad82 Maybe you want to give it a try ?

@Drewsif As you see I like your proposal to use -z and -Z instead of -i :smiley: . As long as there are only two compression tools used that's a nice interface. I initially thought we're going to support four zip tools.
I frankly don't like any implicit behaviour of tools (use pigz if available) and thus keep option -a in the current version. A user has to turn on via option to use parallel compression.

In addition current version allows to override the zip options with environment variables PISHRINK_GZIP and PISHRINK_XZ just in case the default options -f9 (pigz) and -T0 (xz) don't fit the users requirements.

framps, 2020-02-22:
@OmegaSquad82 Would you please confirm my branch works as discussed? I'll then create a PR for PiShrink.

OmegaSquad82, 2020-02-24:
@framps sorry I did not get around to it earlier will do that asap...

framps, 2020-02-24:
@OmegaSquad82 Take your time. I just wanted to know you're still interested in a PR :wink:

OmegaSquad82, 2020-02-27:
Hi @framps,
just a small patch for param checks:

`diff --git a/pishrink.sh b/pishrink.sh
index 8f23190..c8e589d 100755
--- a/pishrink.sh
+++ b/pishrink.sh
@@ -93,7 +93,7 @@ prep=false
 ziptool=""
 required_tools="$REQUIRED_TOOLS"
 
-while getopts ":adhipr:svz" opt; do
+while getopts ":adhipr:svzZ" opt; do
   case "${opt}" in
     a) parallel=true;;
     d) debug=true;;`

I then tested options
- [ ] -vz
- [ ] -vaz
- [ ] -vZ
- [ ] -vaZ

and all worked like a charm. Going to test env vars next...

framps, 2020-02-27:
Great @OmegaSquad82 . Thank you very much for testing and the patch. :+1: The issue is already fixed :wink:

OmegaSquad82, 2020-02-27:
Hi @framps,
I'm glad to help.

Using env vars also works, overriding **any** hard coded option to either gzip, pigz or xz. This is in contradiction to the [README.md](https://github.com/framps/PiShrink/blame/other_zip_tools/README.md#L21) which indicates that env vars are for 'parallel compression'.

BR

framps, 2020-02-27:
Good catch 👍 

This behavior is not intentional but I think we should keep this. Then it's consistent for the parallel and non parallel path and allows to customize the ZIP options also for the non parallel case. 

I think we should just update the readme and are fine. @OmegaSquad82 @Drewsif What's your opinion in this?

OmegaSquad82, 2020-02-29:
Hi @framps,

full ack that users should have the ability to override compressor options anyway. Thx for ur work!

BR

framps, 2020-02-29:
I just cleaned up up the code and updated the README. 

@OmegaSquad82 I would like you to appear as a contributor in the PR given the fact you spent your time on testing the PR :wink: . I suggest to create a PR in my repo, update the README and delete the word parallel (your finding) . I'll merge your PR and the will create a PR in this pishrink repo :smiley:

seaniedan, 2020-05-23:
Hello all, on @Drewsif 's current repo, when I try 
sudo /usr/local/bin/pishrink -vaz /tmp/img.img outfile.img
I get:
pishrink.sh: Shrinking image ...
pishrink.sh: Using pigz on the shrunk image ...
/usr/local/bin/pishrink.sh: line 370: pigz: command not found
pishrink.sh: ERROR occured in line 372: pigz failed with rc 0

...looks like the script needs to test for pigz being present.

framps, 2020-05-23:
@seaniedan Good catch :+1: I fixed this in my branch https://github.com/framps/PiShrink/tree/fix_toolcheck. Please verfiy the fix and I will create a PR.

seaniedan, 2020-05-25:
Vielen Danke! Sehr gut!

framps, 2020-05-25:
Great. Thank you very much for verifying.

closed 2020-06-13 by Drewsif