Drewsif/PiShrink #118: works wonderful, thanks

hanscees, 2019-12-14:
Hi,

pishrink works wonderful:
pishrink.sh v0.1.2
pishrink.sh: Copying idroid-201912-rasp+omv.img to idroid-201912-rasp+omv-shrunk.img......
pishrink.sh: Shrunk idroid-201912-rasp+omv-shrunk.img from 15G to 2,1G...

saves 13G! 
thanks for the script! 

Some observations:
- have to wait wait pretty long before it begins printing anything. So perhaps an idea is to begin with saying somethink like: reading in data, this can take a while

- it could not read in an image with brtfs and asked if it was perhaps ufs. No big deal.

Thanks for the disk-space saver! Also used it to migrate to a bigger sd-card, worked wonderfully!

jayjupdhig, 2021-06-10:
Not really...
https://github.com/Drewsif/PiShrink/issues/199

Drewsif, 2023-02-07:
Years late but thank you for the nice words 😄

closed 2023-02-07 by Drewsif