Drewsif/PiShrink #119: Error while shrinking with raspiBackupWrapper

stm-clemens, 2019-12-16:
Hey,

I try to shrink my weekly raspberry backup with pishrink via raspiBackupWrapper.
Right now it fails with the following error:

```
12-16-2019 09:56:50 --- RBK0010I: ubuntu: raspiBackup.sh V0.6.4.3 (2d927a2) stopped at Mo 16. Dez 09:56:50 CET 2019.
Backup succeeded :-)
pishrink.sh v0.1.2
pishrink.sh: Gatherin data...
md5sum: /tmp/tmp.41jRvuDZjh/etc/rc.local: No such file or directory
Creating new /etc/rc.local
mv: cannot stat '/tmp/tmp.41jRvuDZjh/etc/rc.local': No such file or directory
pishrink.sh: Checking filesystem...
e2fsck: Invalid argument while trying to open /dev/loop4
/dev/loop4:
The superblock could not be read or does not describe a valid ext2/ext3/ext4
filesystem.  If the device is valid and it really contains an ext2/ext3/ext4
filesystem (and not swap or ufs or something else), then the superblock
is corrupt, and you might try running e2fsck with an alternate superblock:
    e2fsck -b 8193 <device>
 or
    e2fsck -b 32768 <device>

pishrink.sh: Filesystem error detected!...
pishrink.sh: Trying to recover corrupted filesystem...
e2fsck 1.45.3 (14-Jul-2019)
e2fsck: Invalid argument while trying to open /dev/loop4

The superblock could not be read or does not describe a valid ext2/ext3/ext4
filesystem.  If the device is valid and it really contains an ext2/ext3/ext4
filesystem (and not swap or ufs or something else), then the superblock
is corrupt, and you might try running e2fsck with an alternate superblock:
    e2fsck -b 8193 <device>
 or
    e2fsck -b 32768 <device>

pishrink.sh: ERROR occured in line 58: Filesystem recoveries failed. Giving up...
```

when I run the script by hand it works just fine.

Im on ubuntu 19.10 64bit raspi4

is there anything I can do to get it working I cant flood my server with 200gb backup files :,D

framps, 2019-12-16:
Hello @stm-clemens ,

> when I run the script by hand it works just fine.

Which script are you talking about? raspiBackupWrapper or pishrink?

stm-clemens, 2019-12-17:
Hey @framps,

pishrink.

framps, 2019-12-17:
Then it's an issue with raspiBackupWrapper and no issue with pishrink. Please create an issue in [the raspiBackup repo](https://github.com/framps/raspiBackup/issues) and close this issue for pishrink :wink:

stm-clemens, 2019-12-17:
Well the thing is. I was running the same script on my raspbian buster installation just fine. I copied the exact same script/config to my ubuntu installation and now pishrink wont run with the error message above. nothing changed within the script itself. thats why I thought it would be an issue with pishrink. I'll try my luck within the raspiBackup repo.

closed 2019-12-17 by stm-clemens

framps, 2019-12-17:
>  thats why I thought it would be an issue with pishrink.

That's why I asked which script you use to shrink your ubuntu image :wink: . If you can shrink the ubuntu image with native pishrink but it fails with raspiBackupWrapper it's an issue with raspiBackupWrapper.