Drewsif/PiShrink #130: PiShrink does not work for PiNN / NOOBs Raspbian: Error: Can't have overlapping partitions

typxxi, 2020-04-13:
For years I had been running my Pi "collection" without trouble but then I ran into a problem when I tried if my Backup Restore proceure still would be working. After years I had run out of 8 SD cards and had purchased 5 or 6 Toshiba last summer. This time "backed image" did no longer fit on my new SD card. 

That was the moment when PiShrink kicked in last thursday and since then I stucked day in day out in an error ... I believed that could be caused by my installation, later believed in some how corrupted SD even it was booting and running still OK. 

First I checked 2 old systems always this way: Create a new image and then "pishrink" it.

I always got errors during pishrink and so I tried it with different new SD cards and failed again. 

Then I created a new SD card from a fresh image to make sure that all of my 3 systems were working properly if I created a clone SD from these fresh images. All of these images were working, I could write them on any of my 8 GB SD cards and these systems booted and ran like the original ... and of cause I tried it on Pi 2 , Pi 3, Pi 4 and on Zero cause I could not believe it.

After 2 days I installed brand new raspbian "master SD" on new SD cards and used again a script to create the live image and to shrink that new image from a new raspbian. Each time I got same or similiar errors.

So I assumed that it might be caused by the PiNN installation I had used since its beginning for a lot of reasons.
I knew that PiNN was based on NOOBs and so I tried to find answers. So I created a new NOOBS raspbian buster system with every code and piece coming from a "different source" then my PiNN before. 

Again: Raspbian Buster installed by Noobs could not be shrinked.
Finally I took a 5 year old archived raspbian image  from the "pre PiNN era", put it on a USB stick, inserted that in a Zero W and this time the PiShrink worked ... and not only once but 3 times before I believed it. 
Then I downloaded a pure Raspbian Buster IMG and created a new system on a SD Card and let my script create a live image and then call PiShrink to shrink the new image. Yes, it worked ...

There must be a system immanent difference between the noobs and pinn raspbian installation cause all of my raspbians caused errors.

It would be great to find a solution cause I need features I get from PiNN / Noobs.

Here is the PhiShrink echo with errros and the success for all who might run into the same issue:

pishrink.sh v0.1.2
pishrink.sh: Copying /media/pi/root/BACKUP/PiNN_ZERO_RASPBIAN_SMALL_BASICS-20200413-213748.img to /media/pi/root/BACKUP/Shrinked/PiNN_ZERO_RASPBIAN_SMALL_BASICS-20200413-213748-COMPRESSED.img......
pishrink.sh: Gathering data...
Creating new /etc/rc.local
pishrink.sh: Checking filesystem...
root: 106585/393216 files (0.2% non-contiguous), 814728/1571840 blocks
resize2fs 1.44.5 (15-Dec-2018)
pishrink.sh: Shrinking filesystem...
resize2fs 1.44.5 (15-Dec-2018)
Resizing the filesystem on /dev/loop0 to 877413 (4k) blocks.
Begin pass 2 (max = 208039)
Relocating blocks             XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
Begin pass 3 (max = 48)
Scanning inode table          XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
Begin pass 4 (max = 8779)
Updating inode references     XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
The filesystem on /dev/loop0 is now 877413 (4k) blocks long.

Error: Can't have overlapping partitions.
pishrink.sh: ERROR occured in line 309: parted failed with rc 0

HERE THE SUCCESS with Raspbian SD created from the Raspberry foundation IMG and not by PiNN or Noobs:

pishrink.sh v0.1.2
pishrink.sh: Gathering data...
Creating new /etc/rc.local
pishrink.sh: Checking filesystem...
rootfs: 147574/443520 files (0.6% non-contiguous), 1253198/1824768 blocks
resize2fs 1.44.5 (15-Dec-2018)
pishrink.sh: Shrinking filesystem...
resize2fs 1.44.5 (15-Dec-2018)
Resizing the filesystem on /dev/loop0 to 1673428 (4k) blocks.
Begin pass 2 (max = 100315)
Relocating blocks             XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
Begin pass 3 (max = 56)
Scanning inode table          XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
The filesystem on /dev/loop0 is now 1673428 (4k) blocks long.

pishrink.sh: Shrinking image...
pishrink.sh: Shrunk /media/pi/64GB/PI4.IMG from 7,3G to 6,7G...

John-Mc, 2020-04-13:
From thePishrink readme file:

> If you are trying to shrink a NOOBS image it will likely fail. This is due to NOOBS partitioning being significantly different than Raspbian's. Hopefully PiShrink will be able to support NOOBS in the near future.

lurch, 2020-04-14:
To prevent other users suffering the same frustration as @typxxi has done, maybe it'd be worth PiShrink attempting to "detect" a NOOBS-partitioned SD-card, and displaying a suitable warning message that PiShrink doesn't (yet) support NOOBS installations; rather than failing with a cryptic "Can't have overlapping partitions." message?

framps, 2020-04-14:
pishrink works fine if you use an image which has the well known Raspbian partitioning. From a documentation point of view  in the README prerequisite section it's stated pishrink has issues with NOOBS images. So everybody who uses pishrink should know about the pishrink restrictions.

From a coding point of view a lot of checks and debug statements may be added in pishrink in order to make the script much more robust, userfriendly and maintainable. But this will take a lot of time and test effort which @Drewsif obviously doesn't want to spend. At the end pishrink will have to have about 80% test and check code and 20% real execution code. I think we have to respect his decision.

Maybe someone is willing to spend this huge effort in a fork.

typxxi, 2020-04-15:
> README

> pishrink works fine if you use an image which has the well known Raspbian partitioning. From a documentation point of view in the README prerequisite section it's stated pishrink has issues with NOOBS images. So everybody who uses pishrink should know about the pishrink restrictions.

Unfortunatelly  not cause you will find dozens of websites explaining the pi-shrink usage but close to none with the note "not for noobs or pinn users" ...

And if I had gotten an error message like "have you been using noobs or pinn to create this OS then  pi-shrink can not support you" ... I would never have started that adventure. And even searching for the issue did not lead me to a hit readme ...

Therefore I had written my steps down in both projects to make sure that other will get behind it far easier than me. 

Also: nowadays many are unaware about how their raspbian had been installed cause they had purchased a noobs one and did the job once and forgot about it for years or they were not aware what noobs is really doing. They might have been happy getting RASPBIAN installed no matter how .... and I would have been one of them back in 2015 when all my SD cards sucked initially. I could not boot ... cause I got the false sd cards over and over again. 

Hope there will be a solution... and now I started my hunt for the smallest 8 and 16 gb cards I can get that are fast and reliable too .  That way I hope to avoid the desaster in case of need for new 8 gb sd cards ... and then I would no longer  need pi-shrink

framps, 2020-04-15:
@lurch Given the current #stayHome policy I spent some time to add a simple check in a [branch of my pishrink fork](https://github.com/framps/PiShrink/tree/check4NOOBs) whether more than 2 partitions are in the image which is true for NOOBs and PiNN images. Following error message will be written now when dealing with a NOOBs image:
```
pishrink.sh v0.1.3
pishrink.sh: ERROR occured in line 138: noobs.img has more than 2 partitions. Note: NOOBs or PiNN images are not supported.
```

If that's what you're looking for I'll create a PR.

framps, 2020-04-16:
Just created a PR. I hope @Drewsif will merge it.

lurch, 2020-04-16:
ping @procount - he's now much more familiar with the partitioning schemes used by various OSes / NOOBS / PINN than I am. (I meant to ping him yesterday but forgot!)

@procount Is detecting more than 2 partitions a reliable indicator of whether an SD card is NOOBS / PINN (and therefore not compatible with PiShrink), or is there a more reliable (but hopefully also simple) test?

BTW @framps NOOBS and PINN are both acronyms, so should be in all-caps and not "NOOBs" or "PiNN" :wink:

framps, 2020-04-17:
> BTW @framps NOOBS and PINN are both acronyms, so should be in all-caps and not "NOOBs" or "PiNN"

Just updated the message :wink:. Let's wait for @procount 's answer on the NOOBS and PINN detection.

procount, 2020-04-17:
Sorry for late response - had a busy day!
I wonder if the question should not really be about detecting NOOBS/PINN formats, but rather about trying to detect the formats that pishrink can or can't deal with (NOOBS/PINN format is just one example of formats that pishrink cannot deal with).

So if PiShrink can only deal with a 2 partition (FAT/EXT4) layout, that is what it should be trying to detect, and reject all other formats. NOOBS/PINN formats have 1 partition (when they are first written) 3 when it has been booted for the first time (boot, extended and settings) and more than 3 whenever any OSes have been installed. The presence of an extended partition is also a reasonable way to specifically detect a NOOBS/PINN format, although it's very crude and can also lead to other false positives.

Detecting more than 2 partitions should therefore be sufficient, but maybe a good idea to reject just one partition too.

However, this will also reject higher multiple partition distros, e.g. Android, which now has 4 partitions. This maybe a good thing if PiShrink only deals with 2 partitions, and inserts specific Raspbian scripts to expand the filesystem. But if PiShrink were to be modified to deal with such distros, this condition would have to be relaxed. In Android's case, the first 3 partitions are fixed sizes and the last is variable, so pishrink could be used to shrink it by squashing the last partition just as it does for Raspbian. But that's something for the future I guess. 

I hope that helps!

framps, 2020-04-17:
> Sorry for late response - had a busy day!

NP. That's daily business :smile: 

> Detecting more than 2 partitions should therefore be sufficient, but maybe a good idea to reject just one partition too.

Good idea. That's what I can I add quickly in the code.

> However, this will also reject higher multiple partition distros, e.g. Android,

I don't think that's the target OS of pishrink. But in order to get rid of false positives I think it makes sense to add an additional option which disables the check - just in case :wink:

framps, 2020-04-18:
I just updated pishrink in the PR #132 to check for exactly 2 partitions and to abort if an extended partition was detected. Option -f will turn off these checks. I also added a check for /etc/rc.local. 

Note: The test images below are just images with partitions but no data.
```
framp@majestix:~/github.com/framps/PiShrink (check4NOOBs)$ for f in *.img; do echo "--- $f ---"; fdisk -l $f; sudo ./pishrink.sh $f; done

--- extpart.img ---
Disk extpart.img: 2 GiB, 2097152000 bytes, 4096000 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0xabd1f042

Device       Boot  Start     End Sectors  Size Id Type
extpart.img1        2048  526335  524288  256M 83 Linux
extpart.img2      526336 4095999 3569664  1.7G  5 Extended
pishrink.sh v0.1.3
pishrink.sh: ERROR occured in line 152: extpart.img has an extended partition. Note: NOOBS or PINN images are not supported.

--- noobs.img ---
Disk noobs.img: 2 GiB, 2097152000 bytes, 4096000 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x9540e8fc

Device     Boot   Start     End Sectors  Size Id Type
noobs.img1         2048  526335  524288  256M 83 Linux
noobs.img2       526336 1550335 1024000  500M 83 Linux
noobs.img3      1550336 2574335 1024000  500M  5 Extended
noobs.img5      1552384 1961983  409600  200M 83 Linux
pishrink.sh v0.1.3
pishrink.sh: ERROR occured in line 148: noobs.img has 4 partitions. Note: NOOBS or PINN images are not supported.

--- onepart.img ---
Disk onepart.img: 2 GiB, 2097152000 bytes, 4096000 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0xe8a0a6ab

Device       Boot Start     End Sectors Size Id Type
onepart.img1       2048 4095999 4093952   2G 83 Linux
pishrink.sh v0.1.3
pishrink.sh: ERROR occured in line 148: onepart.img has 1 partitions. Note: NOOBS or PINN images are not supported.

--- raspian.img ---
Disk raspian.img: 311.4 MiB, 326533632 bytes, 637761 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x193b71d3

Device       Boot  Start    End Sectors  Size Id Type
raspian.img1        2048 526335  524288  256M 83 Linux
raspian.img2      526336 637760  111425 54.4M 83 Linux
pishrink.sh v0.1.3
pishrink.sh: Gathering data...
pishrink.sh: ERROR occured in line 207: /etc/rc.local not found in raspian.img
```

Drewsif, 2020-04-19:
I like the idea of throwing an unsupported error on the unsupported distros. It would be nice if parted would play nice with NOOBs. I'd like to spend some time to actually fix the root issue rather than just yell about it. 

Ideally I'd like to be able to shrink any linux image that has it's last partition as the root partition. So I'll spend some time trying to fix the image before shrinking and if not I'll merge #132 until something else is figured out to support them.

procount, 2020-04-19:
NOOBS uses parted, so what is the issue with it in PiShrink?

If PiShrink assumes a standard Raspbian layout with 2 partitions, then if it resizes /dev/mmcblk0p2 on a NOOBS partition, it will be trying to resize an extended partition. This means any enclosed logical partitions (5,6,7,...n) will overlap with it. the solution here is to resize the logical partitions first, then resize the extended partition to cover the logical partitions, ensuring they are all still within it.

lurch, 2020-04-20:
> Ideally I'd like to be able to shrink any linux image that has it's last partition as the root partition.

NOOBS / PINN allow multiple distros to be installed, so potentially a NOOBS image has multiple root partitions, and I guess it might be nice if all of them could be shrunk? But maybe that's beyond the scope of PiShrink and the Backup feature within PINN is better suited for that? :man_shrugging: 

> I'd like to spend some time to actually fix the root issue rather than just yell about it.

You should have a look at https://github.com/raspberrypi/noobs/wiki/Standalone-partitioning-explained and https://github.com/raspberrypi/noobs/wiki/NOOBS-partitioning-explained if you haven't already done so :slightly_smiling_face:

Drewsif, 2020-06-17:
Just an update, I added some code that should shrink these types of images. Just need to ensure the resize on boot is working, for now its disabled with a warning :D

oberchoo, 2021-05-21:
> Just an update, I added some code that should shrink these types of images. Just need to ensure the resize on boot is working, for now its disabled with a warning :D

Hi @Drewsif, thanks for the hard work. I have been using PiShrink for backup and duplication of microSD cards, an awesome tool! Recently I need to shrink NOOBS, but notice it does not expand the Raspberry Pi OS drive. As per your reply, how can we ensure the resize on boot is working, for now its disabled with a warning. I did get a message of:
"WARNING: PiShrink does not yet support autoexpanding of this type of image"

typxxi, 2021-06-10:
sorry, for being so late back in the game but I first had to recover from an accident which took over a year.
Big thank you for all the investigations and progress achieved 🙂 meanwhile - even a tiny warning helps a lot.

As far as I understand it "should" work but does not work and will show a warning message with expanations. 

But what does this mean to benefit from the "should" mentioned here?

"I added some code that should shrink these types of images. Just need to ensure the resize on boot is working, for now its disabled with a warning :D"

"It should but I disabled it" sounds like does not work at all except it will warn anyone. 
I just want to know if I can give it another try or not, cause in case it does not work not with PiNN at all (even with just 1 additional OS like Pi OS over PiNN). 

Meanwhile my colleagues had also found more and more the general issues that a fully backed up image could be reinstalled if we would have sizwise matching SD cards and not such 8GB with just a 3KB smaller size that will cause issues when writing on "too" small cards. 

That was the core mission I had found usefull, also reduce the NAS load caused by images, that are growing over time cause since the glory 4 GB days now the 8 GB SD cards are disappearing and are replaced by 16 GB which means double or 4 times data volume cause the img is just a clone of a physical SD and even 10% used would lead to a 100% size image. 
On top: if you need to restore an image to a new SD the whole copying process would be a lot faster if we would have to deal with just a 10% size image file (copying from NAS / BACKUP to local PC and then writing the SD Card. 

Thanks a lot for your effort in investigations and improvements 🙂- hope that PiNN and PiShrink become compatible friends at least at the basic level of PiNN with just 1 OS only (instead of unlimited if that is the core issue right now).
I am happy that at least a warning will tell everyone that right now PiNN and NOOBS are not supported to save  time.

Drewsif, 2021-06-11:
The shrinking part works, it is the auto expanding that does not. So as long as you are ok with expanding the partition with `raspi-config` or something else until I test the autoexpansion more it is totally fine to use

deep8787, 2021-08-18:
> The shrinking part works, it is the auto expanding that does not. So as long as you are ok with expanding the partition with `raspi-config` or something else until I test the autoexpansion more it is totally fine to use

So does this mean an image of a PINN setup will shrink down now? And I just have to manually expand the partition after flashing the shrunken image? I hate having 256gb images on my NAS drive....

deep8787, 2021-08-18:
Ok I just noticed the _-s__ option when shrinking. I guess I shall just give it a go :)

closed 2023-02-14 by Drewsif